osqzss/gps-sdr-sim #2: not sure if compatible with hackrf

vsboost, 2015-06-24:
I gave it a go and can see the snr increase on about 11 satellites but get
no location data. I used the rocket.csv file that was in the repository and
the latest ephemeris file.

I let the gps-sdr-sim.exe run for around 3-4 min, file was around 900MB in
length.

not sure if it has something to do with the generated bin file, i believe hackrf requires 8bit signed files.

osqzss, 2015-06-24:
Please try gpssim_hackrf.c with the circular user motion and the sample ephemeris file.
The bin file can be transferred with hackrf_transfer tool as follows:
$ hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0
It is running nicely with my ublox LEA-6T receiver with extra 60dB attenuation.

vsboost, 2015-06-24:
Thank you, this is working now..

closed 2015-06-24 by vsboost