osqzss/gps-sdr-sim #3: Generating ECEF

vsboost, 2015-06-27:
This is not an issue more a question, what are you using to generate the ecef data? are there any programs that are freely available that can do it?
Ill close this once you reply.

Thanks

osqzss, 2015-06-27:
I couldn't find any free software to generate user trajectory, either. There are few commercial products, such as SimGen for LabSat and SimROUTE for Spirent simulator.

osqzss, 2015-06-27:
I have just found out a free NMEA simulator from LabSat:
http://www.labsat.co.uk/index.php/jp/free-gps-nmea-simulator-software

This can be used to generate an ECEF user motion file for gps-sdr-sim.
I will give it a shot.

osqzss, 2015-06-27:
It works nicely! Give a quick look into the tools folder.

vsboost, 2015-06-27:
This works perfectly, thanks for your hard work..

closed 2015-06-27 by vsboost