osqzss/gps-sdr-sim #10: `make test` fails (checksum does not match)

pao, 2015-07-07:
Ubuntu 12.04 system, built a16fd8e54dcf0f9dc8144bffad33dc41378dbd5d from git. Quad-core, and the generator is using OpenMP (which is nice!)

```
@ubuntu:~/gps-sdr-sim$ make test
time ./gps-sdr-sim -e brdc3540.14n -u circle.csv -b 8
User motion data = 3000
xyz =  -3813478.0,   3554276.6,   3662785.2
llh =   35.274016,  137.014864,       100.0
Start Time = 1823:518400.0
01   75.5  19.0
02  278.2   8.6
03   44.8  38.2
06  298.1  43.4
09  144.3  29.1
10  222.8  34.6
11   98.7   7.0
12  326.9   5.0
17  350.8  83.8
20   47.2  49.4
23  106.0  35.7
28  201.9  27.9
32   43.1   2.2
Number of channels = 13
Generating baseband signals...
Time = 300.0
Done!
Process time = 845.140[sec]

real    2m23.036s
user    14m3.674s
sys     0m1.646s
test "$(md5sum gpssim.bin | awk '{print $1}')" == "1a1aa6de6ee7faf58c1544a62aabc900"
make: *** [test] Error 1
```

I don't have the hardware to know if the resulting .bin file is a valid trajectory.

I get the checksum `717f3c820cd62f0d5cd7050be7090f9b` for this build.

pao, 2015-07-07:
I get the same checksum with `OMP_NUM_THREADS=1`, so I don't think it's nondeterminism caused by parallelism. Incidentally, it runs in about 1m20s wall time with 1 OpenMP thread, which is quite a bit faster. This is on a Core i7-4710HQ.

sgk, 2015-07-08:
Hmm. I tested on several machines running Ubuntu 15.04 and 14.04 on x86_64 arch with different minor CPU models. The results are all same MD5 checksum.

According to your result, the difference may be caused by an accuracy of floating point calculation. It may not be appropriate to test the program using checksum.

@osqzss Could you calculate the MD5 checksum of your results on Windows?

osqzss, 2015-07-08:
I got the following result on my Windows environment and the FCIV tool from Microsoft:

```
>gps-sdr-sim.exe -e brdc3540.14n -u circle.csv -b 8
User motion data = 3000
xyz =  -3813478.0,   3554276.6,   3662785.2
llh =   35.274016,  137.014864,       100.0
Start Time = 1823:518400.0
01   75.5  19.0
02  278.2   8.6
03   44.8  38.2
06  298.1  43.4
09  144.3  29.1
10  222.8  34.6
11   98.7   7.0
12  326.9   5.0
17  350.8  83.8
20   47.2  49.4
23  106.0  35.7
28  201.9  27.9
32   43.1   2.2
Number of channels = 13
Generating baseband signals...
Time = 300.0
Done!
Process time = 142.258[sec]

>fciv -md5 gpssim.bin
//
// File Checksum Integrity Verifier version 2.05.
//
df454a9474b5d647f37068309a404bc8 gpssim.bin
```

osqzss, 2015-07-08:
The checksum test has been removed from the Makefile.

closed 2015-07-08 by osqzss