osqzss/gps-sdr-sim #11: OpenMP hurting performance (inverse scaling)

pao, 2015-07-07:
While seeing if #10 was due to some kind of nondeterminism caused by OpenMP, I observed the running time for the test is significantly faster for `OMP_NUM_THREADS=1` than with any increased number of threads, and appears to exhibit inverse scaling. Ubuntu 12.04, GCC 4.6.3. GCC 4.8.2 on the same system gives about the same results.

`OMP_NUM_THREADS` | wall clock time
--------------- | ---------------
1 | 1m12.271s
2 | 1m54.911s
4 | 2m13.468s
8 | 2m26.595s

osqzss, 2015-07-08:
Yes, you are right. I can see the same trend on my Windows environment.
Actually, it runs much faster without OpenMP option.
I am not familiar with OpenMP and possibly doing something wrong.

sgk, 2015-07-08:
Ubuntu 15.04 / VirtualBox 4CPU / Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz

`OMP_NUM_THREADS` | wall clock time
--------------------------------|-------------------
1 | 1m20.86s
2 | 1m30.78s
3 | 1m35.25s
4 | 1m41.25s

sgk, 2015-07-08:
Ubuntu 14.04 / Amazon EC2 t2.micro 1CPU / Intel(R) Xeon(R) CPU E5-2670 v2 @ 2.50GHz

`OMP_NUM_THREADS` | wall clock time
--------------------------------|-------------------
1 | 1m20.90s
2 | 1m21.41s

IvanKor, 2015-07-08:
Microsoft Visual Studio Version 12.0
Visual Studio 2013
OS Win7, CPU    i7-4710HQ

------------- 1 ----------------------
//#pragma omp parallel for private(isamp)

Number of channels = 13
Generating baseband signals...
Time = 299.9
Done!
Process time = 100.379[sec]
CPU Load 12%
------------- end 1 -----------------

------------- 2 ----------------------
#pragma omp parallel for private(isamp)

Number of channels = 13
Generating baseband signals...
Time = 299.9
Done!
Process time = 155.403[sec]   !!!!!!!!!!!!!!
CPU Load 100%   ?????????????????????????
------------- end 2 -----------------

------------- 3  ----------------------
//#pragma omp parallel for private(isamp)
if (!omp_get_dynamic())omp_set_dynamic(1);
#pragma omp parallel for

Number of channels = 13
Generating baseband signals...
Time = 299.9
Done!
Process time = 57.751[sec]
CPU Load 35%
------------- end 3 -----------------

------------- 4  ----------------------
//#pragma omp parallel for private(isamp)
if (!omp_get_dynamic())omp_set_dynamic(1);
#pragma omp parallel for
.
.
.
        } // End of omp parallel for
#pragma omp parallel for

Number of channels = 13
Generating baseband signals...
Time = 299.9
Done!
Process time = 34.179[sec]
CPU Load 80%
------------- end 4 -----------------

osqzss, 2015-07-08:
Hi IvanKor,

Where did you exactly put the second "#progma omp parallel for" in your 4th test?
The 1st option without OpenMP support sill gives me the fastest lap on my PC.

IvanKor, 2015-07-09:
Hi osqzss,
...............
...............
    if (!omp_get_dynamic())omp_set_dynamic(1);  // add this !!!!!!!!!!!!!!!!!
    if (navfile[0]==0)
    {
        printf("GPS ephemeris file is not specified.\n");
        exit(1);
    }
................
...............
...............
         // Properties -> Configuration Properties -> C/C++ -> Language -> Open MP Support -> Yes (/openmp)
#pragma omp parallel for   // add this !!!!!!!!!!!!!!!!!
        for (i=0; i<nsat; i++)
        {
................
...............
...............
        } // End of omp parallel for
#pragma omp parallel for // add this !!!!!!!!!!!!!!!!!   x64, Process time = 34.179[sec] CPU load 80%    
        for (isamp=0; isamp<2*iq_buff_size; isamp++)
        {
**********************************

Thanks for Your hard work

sgk, 2015-07-09:
Ubuntu 14.04 / Amazon EC2 t2.medium 2CPU / Intel(R) Xeon(R) CPU E5-2670 v2 @ 2.50GHz

`OMP_NUM_THREADS` | wall clock time | process time
--------------------------------|--------------------|-----------------
1 | 1m19.745s | 72.167
2 | 0m51.437s | 85.408

sgk, 2015-07-09:
The `private(isamp)` parameter in the `#pragma omp` line should be required. Otherwise, the results would be unpredictable (vary for each run).

osqzss, 2015-07-09:
Hi IvanKor,

Thanks for the information. I can see you running an old one.
The latest code still runs faster without OpenMP on my PC, though.

IvanKor, 2015-07-09:
Yes,
latest code runs faster without OpenMP
only old (first) code work with OpenMP

osqzss, 2015-07-15:
The OpenMP support is disabled by default in the current source code.

closed 2015-07-15 by osqzss