osqzss/gps-sdr-sim #22: limitation to 300 seconds

laf0rge, 2015-08-20:
The code contains some static array of user motion data for 300 seconds at 0.1 Hz rate.  This restricts the use case to very short time frames only.
I would want to modify the code to accept a stream of user motion data (without limit in time). The array would be gone, and instead motion data be read 'as needed' rather than pre-reading + buffering all of it during start-up.
Would you be interested in merging such a patch?

osqzss, 2015-08-21:
Yes, continuous signal generation is very attractive, and that's something I am currently working on. If you already have working code, I live to test it.

laf0rge, 2015-08-21:
I don't yet have code, but would be willing to help.  As you are alrady working on it, I will wait for it to appear in some branch of your repository, to avoid any duplication.

closed 2015-08-22 by osqzss

sinak, 2017-03-15:
Hi there, I'm actually using a static location, and I'm wondering why there's a limit to 300 seconds. Would there be any issues with increasing this?

bitdust, 2017-03-16:
@sinak  
I'm working on the realtime version of gps-sdr-sim and then find the way to modify the limitation for static location mode.
#62

sinak, 2017-03-16:
Ah wonderful, thanks @bidtust. I modified the 300 number manually and recompiled and it seems to work just fine.

Have you seen @osqzss's attempt at real-time for BladeRF? https://github.com/osqzss/bladeGPS