osqzss/gps-sdr-sim #26: Problem with running

mehdideveloper, 2015-12-01:
Hi
I have compiled everything correctly.
Here are my checklist to ensure everything's ok:
1- hackrf_info detects my HackRF
2- I can both receive and transmit with my HackRF (I use a simple FM transmitter in GNU Radio to make sure my HackRF's TX board is working)

I run the gps-sdr-sim program with this command line:
gps-sdr-sim -e brdc3540.14n -u circle.csv -b 16 
I have to use "Ctrl+C" to stop it after a while.
Then I use hackrf_transfer, like your ReadMe.
The problem is, none of my phones' GPS, lock to this (I've got a HTC One M8 and an iPhone)
I also use another SDR to receive this GPS signal, but turning it on or off (running hackrf_transfer or closing it) does not affect my spectrum.

What could be wrong?

Thanks
Mehdi

osqzss, 2015-12-01:
Hi Mehdi,

Please use "-b 8" option for 8-bit I/Q sampling for HackRF.
Enjoy :)

afxmedia, 2015-12-01:
hackrf is 8 bit, you're encoding it to 16 bit.

afxmedia, 2015-12-01:
oops, Mehdi has replied. 
Mehdi, thanks for the great tool you've provided! what is the data format for the CSV motion file? are they also coordinates plus elevation in meters?

mehdideveloper, 2015-12-02:
Thanks for reply, but it does not work. Let's review all steps:
1- I compiled your code both on Windows and Mac OS X and it was successful
2- I run this command:
./gps-sdr-sim -e brdc3540.14n -u circle.csv -b 8
which generates a file called "gpssim.bin" (I let it run for a minute and a filesize about 300MB)
3- hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0
I see  the "TX" LED on HackRF turns on.

The output of my HackRF is connected to the ANT500 antenna (provided with the HackRF)
On another computer (1 meter away) I try to see the spectrum with another SDR, but there is no change in "1.575GHZ" area. Nothing at all.

What could be wrong? My antenna? (on both the HackRF and the receiver SDR, I use a telescopic antenna)

mehdideveloper, 2015-12-02:
Update: I used a simple 6 centimeters long wire as the antenna (both for RX and TX)
Now I see some activity on the spectrum, but still I can't get to spoof my phone!

afxmedia, 2015-12-02:
try it with the latest version of the RINEX file, not the one comes with it.

mehdideveloper, 2015-12-02:
Thank you.
1- I'm using "8-bit"
2- I'm using the latest RINEX file (called "brdc3350.15n" downloaded from http://www.naic.edu/aisr/GPSTEC/Archive/2015/335/)
3- I've tried both simple wire and ANT500 as antenna
4- My phones are set to "GPS only" and they're near the transmitter (less than 1 meter away)
5- I'm using another SDR as a receiver to confirm that my HackRF is transmitting.

But still, my phones do not lock! (I'm using Google Maps. Phone try to lock, but they can't)

Any ideas?

Thanks
Mehdi

osqzss, 2015-12-04:
Hi Mehdi,

You have to set the GPS signal level properly. It is very weak and shouldn't be seen on a spectrum analyzer. The typical signal strength is -130dBm, and it is below the noise floor with the GPS signal bandwidth. You may need 30dB to 40dB extra attenuation. Otherwise, the receiver could be overwhelmed and see the unseen signals due to the cross-correlations.

mehdideveloper, 2015-12-04:
Hi
I'm very near to success, but the phones cannot lock to all (fake) satellite signals.
I played with all gain settings (from 0 to 35) and I tried putting my phone at various distances from the HackRF (from a few centimeters to 5 meters), but still the best I get is this:

http://i.imgur.com/P0WJmSZ.png

http://i.imgur.com/izF0qrf.png

I also connected a 30dB attenuator to the output of HackRF, but that didn't help at all (in fact with attenuator, my phone could not detect the signals at all)
I also used the latest file available: brdc3370.15n

Thank you very much for your time
Mehdi

osqzss, 2015-12-05:
Hi Mehdi,

I am not a big HackRF fun because of its poor reference clock stability. Anyway, here is what I did. 

https://goo.gl/xEcfIO

I used the HackRF telescopic antenna with a 20dB attenuator and put my Zenfone 2 right next to it. The signal level is very unstable, and it takes a while to decode the navigation messages. Looks like they aren't the best set of tools, but still you should be able to spoof the location of your smartphone.

mehdideveloper, 2015-12-05:
Hi 
Thank you. So it seems that my HackRF has a problem!
You used a 20dB attenuator and still the signal strength is about 40.
I'm not using an attenuator, but my signal strength (on the Android app) rarely reaches 30 !
It seems there is something wrong in my RF amplifier.

Thanks again
Mehdi

osqzss, 2015-12-05:
You must put some attenuation. In your case, the receiver apparently sees the satellites which shouldn’t be seen. The SNR value relatively gets smaller because of unnecessary cross-correlations, but that does NOT mean the TX power is lower. You need a proper TX signal level to get appropriate results.

mehdideveloper, 2015-12-05:
Ok, I will try with some attenuators (yesterday that I tried with an attenuator, I couldn't even detect GPS signals on my phone!)

mehdideveloper, 2015-12-05:
Hello again
I thought I should let you know this:
Via playing with frequencies, I thought maybe my HackRF has some frequency drift/offset.
Here's a picture of sending on 1,575,420,000 HZ  :
http://i.imgur.com/rFUYAbM.png

And here's when sending on 1,575,414,000 HZ: 
http://i.imgur.com/JMkG8Yt.png
(This is the closest thing I've managed to get yet, although it still does not lock)

I think my problem lies in software, because the HackRF is not faulty (I have tried sending on a lot of frequencies and it works)

Mehdi

mehdideveloper, 2015-12-05:
Another update:
Michael Ossmann (designer of HackRF) says:
```
That is well within the
expected range.  The crystal on HackRF One is specified to have clock
stability of +/- 20 parts per million.
....
Are you and he using the same Android
hardware and software version?  Was he using a HackRF One with the same
software/firmware version and configuration as yours?
```
I am running HackRF firmware 2015.07.2 and my phones run Android 4.3 and 5.0.1
Can you please let me know what HackRF firmware version you're running?

Thanks
Mehdi

afxmedia, 2015-12-05:
Attenuator shouldn't be a requirement to get the lock. As I'm also using a HackRF One, it works with various antenna I put on it for GPS, including the AT500. Have you tried TX some other signal and see if you get a valid one? like a doorbell or something?

osqzss, I like your nice clear case of HackRF, is that the the Kickstarter version? And would you elaborate a bit more on the reference clock stability issue you have?

mehdideveloper, 2015-12-05:
@afxmedia I have sent FM radio on both VHF frequencies (80-150MHZ) and UHF (1.5GHZ-2GHZ) and successfully received that with another SDR

osqzss, 2015-12-05:
Since the reference clock on HackRF is a simple crystal without any temperature compensation, the frequency offset could be too big for the typical GPS receivers. The 20ppm clock stability could cause 30kHz frequency offset for 1.5GHz signal, and it could be larger than the frequency search range of the receiver.

The revision number of my HackRF board is r2-p1, and the firmware version is 2014.08.1. It is not the Kickstarter version, and I got the acrylic case separately from OSH Stencils.
https://twitter.com/OSHStencils/status/501216026894159874

You can also find the DXF file for the case in the HackRF repository.
https://github.com/mossmann/hackrf/tree/master/hardware/hackrf-one/PlasticCase_CAD

afxmedia, 2015-12-05:
@mehdideveloper I have the latest 2015.07.02

@osqzss thanks for the info!

osqzss, 2015-12-12:
Hi Mehdi,

I found that the latest firmware 2015.07.2 works much better and gives more stable data transfer.
You can also feed a TCXO CMOS output into the CLKIN pin to remove the center frequency offset.

https://goo.gl/yPR8DU

FOX924B works just fine and is available from Digikey.

http://www.digikey.com/product-detail/en/FOX924B-10.000/631-1067-1-ND/1024772

afxmedia, 2015-12-15:
osqzss, that's pretty awesome! could you share the schematic for the TCXO board?

osqzss, 2015-12-16:
The design files are now in the repository.

afxmedia, 2015-12-16:
Thanks! I don't see the cap value, and on the design file it's has spots for 2 caps, i guess you skipped the output one and just the bypass cap for the vcc? maybe 0.1uF for the bypass and 15nF for output? Sorry I'm really not good at this so I need to confirm i'm looking at...

Also, I guess I could even substitute with something better like this? https://www.digikey.com/scripts/DkSearch/dksus.dll?Detail&itemSeq=185975910&uq=635858039123786507 , though I guess the cost vs relative improvement might not be worth it for HackRF?

osqzss, 2015-12-16:
You can find the BOM in the Readme.md. I put a 1uF cap for VCC and leave the output cap open.
The HackRF CLKIN requires LVCMOS level, and FOX924B is good enough for GPS signal simulation.

afxmedia, 2015-12-16:
Ah I see. Perhaps DV75C would work then? 
For your GPS Sim, it worked fine without any external clock. I'm just considering adding one anyway since it should work without adding any bulk or additional power supply.

Speaking of power, I guess you just left it in there all the time during other uses as well? I know it only draws 6mA, but did you ever run into stability just straight from a standard USB 2 port, or need to turn down the gain or/and onboard amplifier for it to be stable?

osqzss, 2015-12-16:
DV75C should work. I rarely use HackRF for my project but haven't seen any stability issue with the USB 2.0 power supply.

afxmedia, 2015-12-16:
great thank you!

closed 2015-12-16 by osqzss

flyinhawk, 2021-12-08:
Hello everyone,
I have a question related to topic mentioned above.
I want to use GPS-SDR-SIM with a newer model of HackRF which does not have an acrylic case allowing you to see PCB.
HackRF that I got can be seen from below:
[image]

In this version of HackRF, I have to supply external CLK signal by using an SMA connector, because I cannot use JTAG connectors due to presence of immovable the outer case. I do not even know this version has JTAG connectors.

CLK installation on previous versions:

[image]

On previous comments a TCXO has been suggested which is out of stock right now:
https://www.digikey.com/en/products/detail/FOX924B-10.000/631-1067-1-ND/1024772

Even if I find a new TCXO having same properties with TCXO shown above, how can I drive an external CLK to HackRF?

Because 10MHz TCXOs have a need of output load of 10pF or 15pF,
and HackRF's external CLK input uses 50OHM SMA.

Has anyone tried using external CLK on this new HackRF?