osqzss/gps-sdr-sim #27: Wrong Date Time

afxmedia, 2015-12-02:
Hi, does the brdc file need to be modified to get the correct time? I've tried to use the original and downloaded the latest from the archive, the location is working great but the time is exact -49 hours of EST(where the simulated gps location is). thanks

osqzss, 2015-12-02:
The simulation start time is set to the time of clock of the first ephemeris set in the RINEX file, and it is defined in the GPS time system.

afxmedia, 2015-12-02:
I see. I guess that means it cannot be easily changed to reflect real time? 
Thanks for providing a great tool btw!

osqzss, 2015-12-02:
You can change the TOC in the ephemeris file by hand. Setting an arbitrary start time in the code is a bit tricky, but it may be in a future release.

closed 2015-12-02 by osqzss