osqzss/gps-sdr-sim #30: GPS Satellites same position?

ericjhorton, 2016-02-04:
I am running gps-sdr-sim with a BladeRF with 60dB attenuator. The satellites that my phone is fixing to are all exactly the satellites that are listed when I run gps-sdr-sim, however when my phone things they are all in the same place and will not lock. SNR levels are pretty good (30-40 depending on distance from antenna).

I tried with both the provided brdc file and a brdc file from today. Also tried using ECEF location near to me currently and the coordinated provided.

ericjhorton, 2016-02-04:
Figured it out never mind. Really impressed with this code btw. Thanks!

closed 2016-02-04 by ericjhorton

santiandres33, 2016-03-31:
Hello, nice you solve it,but now I am having a similar question with relate answer, can you explain your findings please.

Dunax, 2018-07-24:
Hi @ejhort08 , please could you write up a quick sentence on what your resolution was to this issue? I have had a similar issue and I though it was related to the BRDC file been used, that the position information had "moved on" - that a more up to date file was required.