osqzss/gps-sdr-sim #31: Trying to use USRP B100...

HughAndersonNUS, 2016-02-11:
Hi,
I generated my gpssim.bin using this command:
`./gps-sdr-sim -v -e brdc0400.16n -l 1.362334,103.992769,100`
(Using a recent ephemeris, and a location near where I live). 
I am trying to use a USRP B100, with gps-sdr-sim. I can see the satellites on "AndroiTS Test", with signal strengths of 30..32, but the phones do not sync to it at all. The angles are given as 0, and no "tick".
When I run
python2.7 gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 20
I get this:
```
Hughs-MacBook-Pro:gps-sdr-sim-master hugh$ python2.7 gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 20
Mac OS; Clang version 6.1.0 (clang-602.0.53); Boost_105700; UHD_003.008.001-MacPorts-Release

-- USRP-B100 clock control: 10
--   r_counter: 2
--   a_counter: 0
--   b_counter: 20
--   prescaler: 8
--   vco_divider: 5
--   chan_divider: 5
--   vco_rate: 1600.000000MHz
--   chan_rate: 320.000000MHz
--   out_rate: 64.000000MHz
-- 
gr::log :WARN: gr uhd usrp sink0 - Sensor 'lo_locked' failed to lock within timeout on channel 0.

UHD Warning:
    The hardware does not support the requested TX sample rate:
    Target sample rate: 2.500000 MSps
    Actual sample rate: 2.461538 MSps
-- Tune Request: 1575.420000 MHz
--   The RF LO does not support the requested frequency:
--     Requested LO Frequency: 1575.420000 MHz
--     RF LO Result: 1575.421083 MHz
--   Attempted to use the DSP to reach the requested frequency:
--     Desired DSP Frequency: -0.001083 MHz
--     DSP Result: -0.001083 MHz
--   Successfully tuned to 1575.420000 MHz
-- 
Using Volk machine: avx_64_mmx_orc
Press Enter to quit:
```
I wondered if the error in the sampling rate is making the system not work?, and if so is there a fix/workaround?
Cheers Hugh

HughAndersonNUS, 2016-02-11:
I have made a tiny bit of progress, but I am still not there.
 I changed the command to specify "-b 8", and a sample rate of 2416538 Sps:
     `./gps-sdr-sim -v -e brdc0400.16n -s 2461538 -b 8 -l 1.362334,103.992769,100`
And then used a matching command:
     `python2.7 gps-sdr-sim-uhd.py -t gpssim.bin -s 2461538 -x 0`
And I now seem to get yellow, and then green ticked satellites, with some of them showing angles. However, the GPS location does not seem to move from "where I am"...
Cheers Hugh

osqzss, 2016-02-11:
The default sample rate of gps-sdr-sim is 2.6MHz. Please use -s option and set the sampling frequency to 2500000.

With my USRP N210, I can play back the signal using "tx_samples_from_file" tool:

./tx_samples_from_file --file gps.bin --type short --rate 2500000 --freq 1575420000 --gain 0

Would you try this first?

osqzss, 2016-02-11:
I'm afraid that the -s option doesn't support such a fine frequency resolution.

HughAndersonNUS, 2016-02-11:
I tried the suggested command, after creating a new file:
`./gps-sdr-sim -v -e brdc0400.16n -s 2500000 -b 8 -l 1.362334,103.992769,100`
(am I supposed to do -b 8?) ... followed by...
` ./tx_samples_from_file --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0`
But I get a similar sampling rate warning when running:
```
Hughs-MacBook-Pro:gps-sdr-sim-master hugh$ ./tx_samples_from_file --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0
Mac OS; Clang version 7.0.2 (clang-700.1.81); Boost_105700; UHD_003.010.git-119-g42a3eeb6

Creating the usrp device with: ...
-- Loading firmware image: /opt/local/share/uhd/images/usrp_b100_fw.ihx... done
-- USRP-B100 clock control: 10
--   r_counter: 2
--   a_counter: 0
--   b_counter: 20
--   prescaler: 8
--   vco_divider: 5
--   chan_divider: 5
--   vco_rate: 1600.000000MHz
--   chan_rate: 320.000000MHz
--   out_rate: 64.000000MHz
-- 
-- Loading FPGA image: /opt/local/share/uhd/images/usrp_b100_fpga.bin... done
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B100
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: SBXv3 RX
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: SBXv3 TX

Setting TX Rate: 2.500000 Msps...

UHD Warning:
    The hardware does not support the requested TX sample rate:
    Target sample rate: 2.500000 MSps
    Actual sample rate: 2.461538 MSps
Actual TX Rate: 2.461538 Msps...

Setting TX Freq: 1575.420000 MHz...
Actual TX Freq: 1575.420000 MHz...

Setting TX Gain: 0.000000 dB...
Actual TX Gain: 0.000000 dB...

Checking TX: LO: locked ...
U
```
With this I get white bars in AndroiTS, but no ticks, and 0,0...

Cheers Hugh

HughAndersonNUS, 2016-02-11:
Hi,
I understand that the sample rate resolution is not fine, but if I change it I get better results. I have some progress to report... I create a file using this:
./gps-sdr-sim -v -e brdc0420.16n -b 16 -s 2461538 -l 1.362334,103.992769,100
And using this playback command
./tx_samples_from_file --file gpssim.bin --type short --rate 2461538 --freq 1575420000 --gain 0
I get yellow and green signals (15 or so), showing (sometimes) angles (although sometimes it just shows 0,0).
When I do get angles, sometimes I have (say) 10 green (ticked/locked) signals, reporting two angles (123,42) but the GPS says "still searching". The readings do seem to vary - sometimes changing to 0.0 and then back to other values a few minutes later...
Could this all be the failure of the B100 to be able to sample at 2500000?
Cheers Hugh

osqzss, 2016-02-12:
It looks a bit strange that the B100 is unable to support 2.5MHz sampling rate.

Regarding AndroiTS behavior, the Android OS seems to give priority to 
Wi-Fi based location.  Please make your smartphone airplane mode and
see how things go.

HughAndersonNUS, 2016-02-12:
I *think* I understand why the unit does not sample at 2500000 Sps. The hardware can output 64MSps, and if you divide that by 26, you get 2461538 Sps. If you divide by 25 you get 2560000 Sps. 
Cheers Hugh

HughAndersonNUS, 2016-02-12:
OK - I have managed to sample at 2500000 Sps, by setting the master clock rate to 50 MSps instead of it's default 64 MSps. My two commands to generate the gpssim.bin file, and to output them to the USRP B100 are:
```
 ./gps-sdr-sim -v -e brdc0420.16n -b 16 -s 2500000 -l 1.362334,103.992769,100
./tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0 --repeat
```
Two of the phones here have now moved, locating themselves 40km from where I am. Thanks for that.

HughAndersonNUS, 2016-02-12:
Hi Takuji,
Thanks for your help. And thanks for your software :)
It seems that, for the USRP B100, you have to modify the sample rate away from the default.
I have been running it in my room all afternoon, and asking visitors to look at their phones. Before closing this thread, I wondered if you could comment on the following observations:
- I notice that some phones do not seem to lock at all. I have had both Android and iOS devices fail to register the "new" GPS location.
- Some phones take a long time - (up to 20-30 minutes).
- I notice that on my phone, running AndroiTS, that it often claims lots of satellites with angles 0-0, and then a few minutes later, the same satellites with more plausible angles, and then later back to 0-0 - cycling on and on.

Are these results normal? If so - can you explain?
Regards Hugh

osqzss, 2016-02-14:
Hi Hugh,

As I mentioned in an earlier post, some smartphones seem to give 
priority to Wi-Fi or cell station based location and timing over GPS.
They may behave differently if you set them airplane mode.

I don't know the internal process of AndroiTS, but it seems to restart
the GPS signal acquisition if the GPS based location and timing are
totally different from the current ones. So you can see all the satellite
angles go back to 0.0 repeatedly.

closed 2016-02-22 by osqzss