osqzss/gps-sdr-sim #32: I want to something about gps signal record in bladeRF.

KIMHYUN-JIN, 2016-02-21:
Hello, everyone.
I am graduate student and I am very interested in GPS security. I will study the gps record&replay, spoofing, countermeasure. Firstly, I will implement a gps record&replay. So, I buy the BladeRFx115, bias-tee, LNA, gps active antenna like unicorn team in DEF CON 23 (https://www.youtube.com/watch?v=jwJKMti_aw0)
I connected a tools like this to record the GPS signal.
<Antenna == Bias-Tee == LNA == BladeRF == Notebook>
But, I could't received the gps signal enough to positioning. My rx is average I/Q= 130, 122.
So, I think that the active antenna isn't operate well. After that, I connected the USB-DC cable to Bias-Tee to energize a DC to the antenna. Therefore, I get the I/Q = 270,  224. And I tested the rx-tx using this command and configures.
1) rx command : rx config file=gpssim.bin format=bin n=20M
2) tx command : tx config file=gpssim.bin format=bin 
3) configure
set frequency 1575.42M
set samplerate 2M
set bandwidth 2M
set config format=csv n=8M

So, how can I solve this problem and I want to know about your tips please .
thank you.

osqzss, 2016-02-22:
Please try the steps on my blog post: http://blog.goo.ne.jp/osqzss/e/b9949a5e51707c4a89f73b520b07a9c9

KIMHYUN-JIN, 2016-02-24:
I'm sorry for late answer.

I tried to test like your post.

But, I think that rx is the problem..

This is my testing history : http://blog.naver.com/be_successor/220637262557 

I am waiting for your answer

Thank you.

osqzss, 2016-02-24:
ZX60 is not a bias-T. There is no DC out from IN port. You have to feed DC to the active antenna.

closed 2016-03-11 by osqzss