osqzss/gps-sdr-sim #33: Don't work with Hackrf one

oldxfile, 2016-02-22:
I download and tried these great codes. But encountered some problem while trying to send signals through Hackrf one. I generated gpssim.bin through 
**./gps-sdr-sim -b 8 -e brdc3540.14n -u circle.csv**
(I also tried with fixed location -l option and latest brdc file, results were the same)
Then I tried to let Hackrf to send the signal:
**sudo hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0**
(I tried it on both Mac and a Linux virtual machine, results were the same)

Before I start hackrf's transfer, my android phone caught some gps signal leaked from the window: 
[image]

After I start Hackrf's transfer, in about 30 seconds, I got this:
[image]
No any bars at all! The Hackrf looks like a signal jammer here.

When I stop the Hackrf's transfer, the bars were back:
[image]

The Hackrf one was using the antenna came with it. The phone running AndroiTS was located about 0.4 meter from the antenna. 
Did anyone encounter the same problem?

osqzss, 2016-02-22:
Possibly an unstable crystal on your HackRF.
Please check the following thread: https://github.com/osqzss/gps-sdr-sim/issues/26

oldxfile, 2016-02-22:
OK. I'll try to attach a FOX924B-10.000 to it and come back later.
Thanks.

osqzss, 2016-02-22:
The TCXO is helpful but not required. 
You can first try adding some offset to the TX frequency like Mehdi did.

oldxfile, 2016-02-23:
Oh, I've already tried different TX frequencies but got no luck. Will see what will happen after I get ext clock ready.

oldxfile, 2016-02-26:
I've tried ext clock. A FOX924B-10.000 didn't help at all. The results were exactly the same. Meanwhile, I double checked Hackrf one's TX. I built a simple FM broadcasting station through GNU Ratio. The Hackrf board works perfectly transmitting WBFM signals. 
   Any suggestions?

osqzss, 2016-02-26:
Would you try the bit stream file (circle_hackrf.bin) in the following folder?
https://goo.gl/m8OpwZ

It was generated in my environment and works fine with my HackRF One.

oldxfile, 2016-02-26:
I'll download it and have a try. Thanks a lot!
Oh...It's a 1.5GB file...And I need to download it from GOOGLE.

oldxfile, 2016-02-26:
Latest update: It works! It's not the data file, but the computer! I changed to a windows laptop and it works immediately. It looks that something wrong with my Macbook pro or its USB port.
Thanks again for your kindly help.

closed 2016-02-26 by oldxfile

oldxfile, 2016-02-26:
And the external clock is a must. My Hackrf one failed while using its own clock.

dasdboot, 2016-03-19:
hackrf internal clock is not stable enough for most of the GPS receiving units, to make sure this variable is out of the equation make sure you drive the hackrf from 10mhz with less then 2.5ppm drift.
http://community.hpe.com/t5/Security-Research/Spoofing-GPS-for-the-budget-conscious/ba-p/6838676

DL7AD, 2016-03-19:
I use the gps-sim almost every day for automatic testing. The hackrf produces a reasonable signal with it's internal oscillator. But it works much much better with an external reference. I have a 0.5ppm reference for this purpose.