osqzss/gps-sdr-sim #34: What is the format of the csv file

nzkarit, 2016-03-11:
Hi,

I'm looking at the circle.csv file and was wondering what each column was and what format it is in. Is there any documentation for it anywhere? I would like to make my own shape.

Thanks

osqzss, 2016-03-11:
The CSV file needs to contain 10Hz user motion in the ECEF frame. The time stamps in the first column are not used in the simulator. You can use the llh2xyz function in the main code to transform your geographic position into the ECEF.

closed 2016-03-11 by nzkarit

szakacsm, 2020-02-09:
Hi, if I have a user motion file in kml format how can I transform it to ECEF frame?

osqzss, 2020-02-09:
Use llh2xyz function in gpssim.c.

szakacsm, 2020-02-10:
in visual studio? how can I give the input (user motion) kml file? 
can you please give detailed explanation?
I am not an expert programming guy

szakacsm, 2020-02-12:
I am getting to understand what you said
Please confirm the following
llh[0] = latitude
llh[1] = longitude
llh[2] = ???

osqzss, 2020-02-12:
height