osqzss/gps-sdr-sim #35: SIGNAL STRENGTH

santiandres33, 2016-03-31:
Hello,

First of all, bright software tks. 

I have made it work fine with the BladeRF40x. 

However, I dont know why my SNR values (signal strength) is that different for each satellite(between 41-50 for twelve fixed satellites), and it is not the same value as it should be, considering this is an emulation-simulation of satellites, so I would think that the strength must be the same because it is only one source (In all the pictures and screenshots I have seen, the values are for ex. 42-44 for eighth sats or 39 for eleven sats)

Is it possible to adjust this SNR values??

FIG. 5 [https://www.blackhat.com/docs/eu-15/materials/eu-15-Kang-Is-Your-Timespace-Safe-Time-And-Position-Spoofing-Opensourcely-wp.pdf](url)

SLIDE 42 and 43 [https://media.defcon.org/DEF%20CON%2023/DEF%20CON%2023%20presentations/DEFCON-23-Lin-Huang-Qing-Yang-GPS-Spoofing.pdf](url)

osqzss, 2016-03-31:
The simulator applies different pass-loss and receiver antenna gain to each signal.
Set all the gain[i] in line 1957 to 100 then you will get the same signal strength for
all the satellites.

santiandres33, 2016-04-01:
Thanks for your soon response.

I changed this line `gain[i] = (int)(path_loss*ant_gain*100.0)` for `gain[i] = 100` in the file gpssim.c. But I still saw different SNR values for the satellites, so do I this values can be defined? 

Where can it be managed the signal strength generated by the SDR (bladeRF)?

osqzss, 2016-04-01:
The gain[i] is the only parameter to alter the relative gain among the satellites.
If you set it a constant, all the SNR values should be the same.

The absolute TX power of the bladeRF can be managed by txvga1 to adjust
the amount of attenuation.

santiandres33, 2016-04-03:
Hello,

Yes, by changing this parameter (`gain[i] = 100`) the values of SNR for each satellite, are similar, there is slightly variations from 1 to 2. Interesting thing, some GPS receivers don't fix any satellite when the SNR values are that similar and that high.

santiandres33, 2016-04-03:
txvga1 is set to -25 in the bladerf script, this means that the strength of the would be 25db?

osqzss, 2016-04-06:
No, it applies 25dB attenuation to the absolute TX power.
The typical TX power is said to be 6dBm, but I don't know the exact value at GPS frequency.
https://www.nuand.com/blog/product/bladerf-x115/

santiandres33, 2016-04-06:
Understood, again bright software. tks.

closed 2016-04-06 by osqzss