osqzss/gps-sdr-sim #36: Wrote a parsing method for LLH CSV

nzkarit, 2016-04-14:
Hi,

I have written a method that will take a lat, long, height csv and generate the file. It is based on NEMA method but designed for the following CSV:
```
0.1,41.259519,174.813354,100
0.2,41.259519,174.813354,100
0.3,41.259519,174.813354,100
0.4,41.259519,174.813354,100
0.5,41.259519,174.813354,100
0.6,41.259519,174.813354,100
0.7,41.259519,174.813354,100
0.8,41.259519,174.813354,100
0.9,41.259519,174.813354,100
1,41.259519,174.813354,100
```
Time, Latitude, Longitude, Height

It would be great if you could include the following in the application:
```C
int readLLH(double xyz[USER_MOTION_SIZE][3], const char *filename)
{

	FILE *fp;
	int numd = 0;
	char str[MAX_CHAR];
	char *token;
	double llh[3],pos[3];
	char tmp[8];

	if (NULL==(fp=fopen(filename,"rt")))
		return(-1);

	while (1)
	{
		if (fgets(str, MAX_CHAR, fp)==NULL)
			break;

		token = strtok(str, ","); //Take out leading time

		token = strtok(NULL, ","); // Latitude
		strncpy(tmp, token, 2);
		tmp[2] = 0;
		
		llh[0] = atof(tmp) + atof(token+2)/60.0;

		llh[0] /= R2D; // in radian

		token = strtok(NULL, ","); // Longitude
		strncpy(tmp, token, 3);
		tmp[3] = 0;
		
		llh[1] = atof(tmp) + atof(token+3)/60.0;

		llh[1] /= R2D; // in radian

		token = strtok(NULL, ","); // Altitude above mean sea level
		
		llh[2] = atof(token);

		// Convert geodetic position into ECEF coordinates
		llh2xyz(llh, pos);

		xyz[numd][0] = pos[0];
		xyz[numd][1] = pos[1];
		xyz[numd][2] = pos[2];
		
		// Update the number of track points
		numd++;

		if (numd>=USER_MOTION_SIZE)
			break;

	}

	fclose(fp);

	return (numd);
}
```
One thing I have noticed with a Lat of "41.259519" comes up correctly yet "-41.259519" is coming up a 3 deg North. Not sure where that is happening? Maybe the llh2xyz method?

osqzss, 2016-04-15:
Could you give me a bit more details on your "3 deg North" issue? When you convert the negative latitude into XYZ and then covert back to the geodetic coordinate, is the resulting latitude 3-deg higher than the original one? Or does your GPS receiver give you a wrong location (3-deg is a huge error!) when applying a negative latitude to gps-sdr-sim?

nzkarit, 2016-04-15:
My GPS receiver is showing 3 deg north (i.e. north of the equator) when provide it with -41.259519.

I am not doing the converting, I am using the existing llh2xyz method which is in the code.

osqzss, 2016-04-15:
In your code, you only copy the first two characters from the latitude token. If you apply -41.259519 for the latitude, the temp string can contain only "-4", not "-41". By the way, you don't need to divide the fractional part of the latitude and longitude by 60.0 since they are already given in degree.

nzkarit, 2016-04-15:
Thanks that will be it cheers. My C is a tad rusty.

closed 2016-04-29 by osqzss

apppie123, 2023-04-08:
I've had some trouble on getting a fix on some GPS receivers using LLH format in a CSV file.

When I put a value for height to 3 meter or 8 meter, one of the devices would not get a fix.
It would sometimes get a fix, but then lose it directly or no fix at all.

The height represents meter, but what's the reference or zero point? Is there a minimum height value for it work?