osqzss/gps-sdr-sim #38: GNSS Support

davidscott39, 2016-07-19:
Hello,

Have you considered adding support for GNSS to gps-sdr-sim to include the option for realtime GPS data?

http://gnss-sdr.org/

Otherwise, does anyone know how you might precompute the brdc rinex files for future times?

Thanks!

osqzss, 2016-07-20:
There's no plan at this point, but any contributions, but any contributions are always welcomed!

zimler, 2016-07-28:
@davidscott39 have you looked into the Ultra-Rapid(predictive half) data type? It _seems_ like that is what that product is - a predictive file, but I'm not sure if it is available ahead of time or not... https://igscb.jpl.nasa.gov/components/prods.html

I'm trying to get real time GPS transmission, too.

osqzss, 2016-07-31:
I added the option '-T' to overwrite the TOC and TOC in the RINEX navigation file to the user-specified scenario start time. See if it works.

zimler, 2016-08-01:
I can try tonight, in about 8 hours. Thank you!

zimler, 2016-08-02:
@osqzss the -T performed as expected. Thanks!

closed 2016-08-09 by osqzss

zxcvbngg, 2018-05-04:
666