osqzss/gps-sdr-sim #41: interactive mode

FrankBuss, 2016-07-22:
Would be nice to have an interactive mode, where you can enter (or pipe) the latitude/longitude/height with stdin. I hacked this feature in the existing version, see here: https://github.com/FrankBuss/gps-sdr-sim , where you can click with the mouse on Google Maps and then it should send the new GPS coordinate. I tried to test it, but the GPS receiver doesn't like my HackRF, it works very seldom. I ordered a better internal TCXO crystal from eBay.

Maybe someone can test it and if it works, I can cleanup the source code to integrate it more nicely in the existing app.

Usage:

- compile the modified gps-sdr-sim app:
```
gcc gpssim.c -lm -lpthread -O3 -o gps-sdr-sim
```

- download the latest GPS broadcast ephemers file (brdc) from 
ftp://cddis.gsfc.nasa.gov/gnss/data/daily/ , e.g.:
```
wget ftp://cddis.gsfc.nasa.gov/gnss/data/daily/2016/brdc/brdc2040.16n.Z
gzip -d brdc2040.16n.Z
```

- create a FIFO:
```
mkfifo gpssim.bin
```

- start the nodejs app, piping the coordinate output to the (modified) gps-srd-sim, which writes the SDR data to the FIFO (you might need to install nodejs first with ```apt-get install nodejs``` and then socket.io with ```npm install socket.io```) :
```
nodejs maps/app.js | ./gps-sdr-sim -b 8 -e brdc2040.16n -l 50.9413,6.9583,157
```

- open another terminal and start the HackRF (or other SDR) transfer program:
```
hackrf_transfer -t gpssim.bin -f 1575423000 -s 2600000 -a 1 -x 0
```

- open the address "http://localhost:7000/index.html" in a web browser. You should see Google maps, centered around the Cologne Cathedral. Click with the mouse to the position where you want to go. Each mouse click should be displayed in the nodejs application, e.g.
```
new coordinate: 50.935397 6.960944 157.100000
```
The third parameter is fixed and the height. The data is streamed to the FIFO and the modified GPS signal is sent continuously.

PS: works only on fast computers where the signal generation is faster than real-time.

osqzss, 2016-07-23:
Thank you for the input! The interactive mode is certainly attractive, but I hesitate to add this capability to the master repository because the named pipe is not available in Windows. I would rather keep it a simple baseband generator.

By the way, the gps-sdr-sim calculates the user velocity from two consecutive locations every 0.1 seconds. You cannot simply select any location for the next time step. It would generate unrealistic user dynamics, and the receiver could never catch the signal. You need a path generator, but it is not trivial.

FrankBuss, 2016-07-23:
Named pipes are available in Windows ( https://msdn.microsoft.com/en-us/library/windows/desktop/aa365590(v=vs.85).aspx ), but not as widely used as in Unix, and there is no command line tool to create it. But you are right, maybe not a good idea to integrate it to the already very big program, and pthread which I used to monitor stdin in a second thread would be another problem in Windows.

Maybe a better idea would be to refactor the core calculating parts of your program to a library? This would make it easier to use it from other programs. Then the interactive program could be just a small extra program, with less arguments, which might only work in Linux (or maybe it does with the new Linux subsystem in Windows, which should allow to run just any Linux program native on Windows, http://betanews.com/2016/04/24/windows-10-linux-subsystem/ ), but the simple baseband generator program would still compile for any platform.

Independently of this, adding support for stdout might be a good idea, then you don't need the temporary bin file, because I think hackrf_transfer can already use stdin.

Good idea with the path generator, shouldn't be too difficult.

osqzss, 2016-07-23:
That sounds intriguing, but it is clearly beyond my ability. I've never been a good programmer, and my first attempt for a real-time simulator is a quite mess...
https://github.com/osqzss/bladeGPS

It would be nice if we could have a simple baseband generator library portable to many other platforms. Maybe GNURadio?

ehad16, 2016-08-14:
@osqzss, how hard would it be to make that 'wasd' version work with the hackrf?  Where do you specify how much of a jump that 'wasd' can do?

osqzss, 2016-08-15:
With a constant acceleration and a fixed maximum velocity, the path generation with the 'wasd' keys shouldn't be that difficult. Many GPS receivers can keep signal lock under <2G dynamics environment.

DL7AD, 2016-08-15:
This is so coool! Thanks a lot. It worked on my phone very well. Jumping around Cologne (up to 5km) worked well. The GPS inside my mobile got the new fix within 5 seconds. Jumping (above ~5 km) between Cologne and Berlin (500km) caused a GPS loss which was reacquired after ~40 seconds.

Out of my memory I know that my uBlox test modules don't like jumping around the world. They can get confused quite a lot. Sometimes I have to reset them to get it working again.

You just did a minor error in your description. The frequency is 1575420000 not 1575423000 (Hz). You maybe set this frequency accurately for your device, but while I have my SDR connected to a 100ppb reference, it will be offset.

osqzss, 2016-08-20:
Check this out:
https://www.youtube.com/watch?v=ftjlT7-ErCQ

ehad16, 2016-08-21:
thats pretty awesome, any chance for that to come available?

osqzss, 2016-08-21:
WALB is a HackRF + Raspberry Pi tool developed by Keiichi Horiai, and it was presented at Black Hat USA 2016 Arsenal.
https://www.blackhat.com/us-16/arsenal.html#keiichi-horiai

Hope it will be available in his GitHub repository:
https://github.com/crescentvenus/WALB

ehad16, 2016-08-22:
Doesn't look like he has interest in sharing

KM4TXM, 2016-08-23:
I've been trying to do something similar to this, and having no luck,

I have downloaded the ephemeris from NASA, built the gpssim.bin file, and
run the program to create a GPS signal.

I connected it to a GPS receiver via an SMA cable and 50dB attenuator from
a HackRF One. I get no information on the receiver at all.

I connected two HackRF One radios together, one with the gps-sdr-sim
software running, the other with a FFT sink watching 1.5GHz -- and I see
nothing.

It's as if there gps-sdr-sim software, though running, accomplishes nothing.

Any thoughts?

./kevin

- Kevin

osqzss, 2016-08-23:
This could be some help: https://github.com/osqzss/gps-sdr-sim/issues/33

closed 2016-11-18 by osqzss