osqzss/gps-sdr-sim #42: Testing leapsecond issues

vik976, 2016-07-29:
The software is working nicely with HackRF + ext sync - tested with 4 different GPS receiver, but I would like to test equipments for the previous 30.06.2015 and the upcoming 31.12.2016 **leapseconds**.
Even with the correct brdc1810.15n ephemeris file I don't get the leap second on- which is natural since the generator is not handling this - but it is beyond my ability to find the right place in the code to add. 

For upcoming leapsecond I need a helping idea, how to generate files for the future.
Every good answer is appreciated!

osqzss, 2016-07-30:
You need to add the UTC parameters to the navigation message. Please refer to the section 20.3.3.5.2.4 Universal Coordinated Time of ICD-GPS-200C for details:
http://www.gps.gov/technical/icwg/ICD-GPS-200C.pdf

Page 18 of subframe 4 includes (1) the parameters needed to relate GPS time to UTC, and (2) notice to the user regarding the scheduled future value of the delta time due to leap seconds.

osqzss, 2016-08-03:
The current and future leap seconds parameters are now  supported. Let' me know if it works for you.

vik976, 2016-08-03:
First of all, thank you for the fantastic work. I am trying to understand how to use the -t and -T options in order to generate the next leap second that will occur in 31.12.2016. I think that an example would save me from lot a pain. Many thanks again.

nzkarit, 2016-08-16:
@vik976 I am interested in knowing this as well. I tried editing gpssim.c with the dtlsf variable which appears to be the number of leap seconds. I set it to 600 to make it clear it was having an effect but after recompiling it didn't seem to have an effect :( Nor did change the rinex file :( There is a TODO in the code suggesting that leap seconds will be brought out as a command line in the future, so fingers crossed. C isn't a language I know particularly well so I have issues understanding what is going in the code.

osqzss, 2016-08-16:
Both the dtls and dtlsf are 8-bit signed integers. You cannot set them to 600 (or it will be 88). The dtlsf is the scheduled "future" value. It will be effective when the receiver time falls in the time span of DN+3/4 to DN+5/4 days. You would like to change the "current" leap second, edit the LEAP SECONDS parameter in the RINEX file.

osqzss, 2016-08-16:
There're some errors in week number transition and leap second handling. I've just uploaded the revised version, and it can handle the leap second nicely :)

Scheduled leap second:
[image]

Leap second insertion (2016/12/31 23:59:60):
[image]

closed 2016-09-01 by osqzss