osqzss/gps-sdr-sim #43: latest build fails

crispyjones, 2016-08-06:
Build fails with following warnings on 14.04 and 16.04
fails on a machine with older working install 

gpssim.c: In function ‘allocateChannel’:
gpssim.c:1580:33: warning: passing argument 3 of ‘eph2sbf’ from incompatible pointer type [-Wincompatible-pointer-types]
       eph2sbf(eph[sv], ionoutc, chan[i].sbf);
                                 ^
gpssim.c:491:6: note: expected ‘long int (*)[10]’ but argument is of type ‘long unsigned int (*)[10]’
 void eph2sbf(const ephem_t eph, const ionoutc_t ionoutc, long sbf[5][N_DWRD_SBF])

osqzss, 2016-08-06:
Does the latest source resolve the issue?

crispyjones, 2016-08-07:
Yes, builds without warnings as expected, thank you.

closed 2016-08-07 by osqzss