osqzss/gps-sdr-sim #45: Error with my usrp n210

xiaof3i, 2016-08-09:
i run gps-sdr-sim on ubuntu 14.04 64bit
when i generate the gpssim.bin, 
gps-sdr-sim -e brdc3540.14n -l 30.286502,120.032669,100  -s 2500000 -b 8
every thing seems work fine

However, when i execute
gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0
it throw a error
it says

what's wrong with that?
can anyone help me？

DL7AD, 2016-08-10:
Hi,

Ubuntu 14.04 runs an old version of gnuradio. (I don't how old).
But I know, they have changed the arguments. I run Ubuntu 16.04 which has gnuradio 3.7.10.

Looking at the doc of gnuradio 3.6.5 it did not have those arguments.
v3.6 http://gnuradio.org/doc/doxygen-3.6/interleaved__short__to__complex_8h_source.html
v3.7 http://gnuradio.org/doc/doxygen-3.7/interleaved__short__to__complex_8h_source.html

So you could either try to remove both arguments in line 41 (at make) or to update gnuradio. The first option would be the quickest. But you could also run into other compatibility problems.

xiaof3i, 2016-08-15:
thx~ it seems you are right
it is all because a old version of gnuradio

closed 2016-08-15 by osqzss

donghuixin, 2019-04-28:
A similar error occurs on my USRP X300, but my version of gnuradio is 3.7.13.4。
Also when I generate the gpssim.bin,
everything seems to work fine
when I execute
gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0
it throws an error says:
[ERROR] [UHD] Exception caught in safe-call.
  in ctrl_iface_impl<_endianness>::~ctrl_iface_impl() [with uhd::endianness_t _endianness = (uhd::endianness_t)0u]
  at /home/zxn/workarea-uhd/uhd/host/lib/rfnoc/ctrl_iface.cpp:52
this->send_cmd_pkt(0, 0, true); -> EnvironmentError: IOError: Block ctrl (CE_00_Port_30) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0u; uint64_t = long unsigned int]
  at /home/zxn/workarea-uhd/uhd/host/lib/rfnoc/ctrl_iface.cpp:142

Traceback (most recent call last):
  File "gps-sdr-sim-uhd.py", line 90, in <module>
    tb = top_block(options)
  File "gps-sdr-sim-uhd.py", line 23, in __init__
    self.uhd_usrp_sink = uhd.usrp_sink(options.args, uhd.io_type_t.COMPLEX_FLOAT32, 1)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/__init__.py", line 122, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 3127, in make
    return _uhd_swig.usrp_sink_make(*args)
RuntimeError: EnvironmentError: IOError: Block ctrl (CE_00_Port_30) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0u; uint64_t = long unsigned int]
  at /home/zxn/workarea-uhd/uhd/host/lib/rfnoc/ctrl_iface.cpp:142
can anybody tell me how can i fix it?