osqzss/gps-sdr-sim #46: Bug about stimulating GPS timing

xiaof3i, 2016-08-16:
hello, i use this project to stimulate the process of GPS timing for my device.
I used the brdc in the project and i successfully set my time to 2014/12/20
however, i download a brdc file of 1997. Bug encoutered. I found that the time was set to 2097

So i read the sourcecode of gpssim.c
i found this in LINE 936
`
		// EPOCH
		strncpy(tmp, str+3, 2);
		tmp[2] = 0;
		t.y = atoi(tmp) + 2000;
`

so I change the 2000 to 1900 in order to stimulate the times in 1900

but it seems dont work?

Can anyone help me?

xiaof3i, 2016-08-16:
Is it possible for me to set time to 1969 or 1970
i modify the Line 185 `ye = t->y - 1980;`
to ye = t->y - 1950;
to avoid a negative value. 
Will it influence the timing?

osqzss, 2016-08-17:
No. GPS time zero started at 00:00:00 on January 6, 1980.

xiaofen9, 2016-08-18:
can i achieve my goal of 1970 by modify your code?
when i write like this?
ye = t->y - 1950;

xiaof3i, 2016-08-20:
Can i stimulate a time in the future?
Like 2020?

osqzss, 2016-08-20:
Try "-T" optoin. 
Some more discussion on the simulation start time can be found here:
https://github.com/osqzss/bladeGPS/issues/14

xiaofen9, 2016-08-21:
Thank you for offering so much help and thank you for your project. It really helps me.
after using -T  option
i can stimulate the time in the future.
But i still get some problem
When i use -T 2032/01/01,11:11:11 , it works really fine！ I can see the utc time has been changed to 01-01-32 in my android app GPS TEST PLUS.  the time before 2032 works well as welll

**HOWEVER**  when i try to stimulate time after that, 
for instantce
-T  2033/01/01,11:11:11 it changes to 01-01-13
and so on
-T  2037/01/01,11:11:11 =>  01-01-17
-T  2048/01/01,11:11:11 =>  01-01-28

My aim is to test the GPS timing function of an ARM devices, so i need to brocast info of years in a wild range(from 1980~2039)

Will you fix the problem, it will help me from desperate! thank you!

osqzss, 2016-08-21:
It's not the issue of the simulator but the way to handle the week number rollover in your receiver. 
My u-blox NEO-M8N works just fine even on Jan. 01, 2039.

[image]

xiaofen9, 2016-08-22:
Got it！
you means that my device "suffers" from a rollover!
It is quite awkward right?   : )
SO it is related to the firmware.... and i can do nothing with that ?  :(

closed 2016-09-01 by osqzss