osqzss/gps-sdr-sim #47: Trouble locking in with HackRF

ehad16, 2016-08-29:
I am getting high 40s and some 50s, but I can't seem to lock in with the hackrf.  Tried a few different phones with no success.  LG G2, iPhone 5s, nexus 5, and iPhone 6s Plus.  I am using a vat30 and the small orange antenna from digikey.  Have tried the ant500 as well.  I am using this tcxo from eBay.  http://m.ebay.com/itm/HackRF-one-external-TCXO-clock-PPM-0-1-for-GPS-Applications-GSM-WCDMA-LTE-/272320312607?txnId=1688277940017

Getting 0x01 with it in.  Any ideas?  I am using -b 8.

osqzss, 2016-09-05:
I hope this will be of some help:
https://nuand.com/forums/viewtopic.php?f=4&t=3881

osqzss, 2016-09-14:
GPS Spoofing w/ BladeRF - Software Defined Radio Series #23 
https://www.youtube.com/watch?v=VAmbWwAPZZo

closed 2016-09-18 by osqzss