osqzss/gps-sdr-sim #48: Data inconsistency issue

CollusW, 2016-09-06:
Hi,

I met some data inconsistency issues when I tried to ''catenate“ different IQ waveforms. In one word, the waveform that is generated for certern duration is not equal to the waveform that is generated and catenated for the portion of that duration.
(I think it is a useful feature to this project because it enables the program to generate waveforms for infinite duration, not limited to memory size.)

Here is what I do and some analysis:

For example, I generate two bin files (waveform) and each duration is 1 second, using the following commands:
 
```
GpsSdrSim.exe  -d 1.1 -e .\ephemerides\brdc0010.15n -b 8 -l 18.227617,109.524737,100.000000 -s 1023000 -T 2016/11/01,23:59:30 -o gpssim_1.bin
GpsSdrSim.exe  -d 1.1 -e .\ephemerides\brdc0010.15n -b 8 -l 18.227617,109.524737,100.000000 -s 1023000 -T 2016/11/01,23:59:31 -o gpssim_2.bin
```
 
Notice that the waveforms should seamlessly cover the time from 30'' to 31'' for 2 seconds.

Then I generate another bin file whose duration is 2 second, same start time, using the following commands:
 
```GpsSdrSim.exe  -d 2.1 -e .\ephemerides\brdc0010.15n -b 8 -l 18.227617,109.524737,100.000000 -s 1023000 -T 2016/11/01,23:59:30 -o gpssim_1and2.bin```

Then I compare (abs(difference)) the catenated one [gpssim_1.bin, gpssim_2.bin] with gpssim_1and2.bin. I found that the first second waveforms are the same but the last second waveforms are different as you can see from the matlab figure below.
 
[image]

Some further analysis:

I did cross-correlation between the 1s waveforms and the 2s waveform. 
Xcorr between the first 1s waveform and the 2s waveform
[image]
As expected, the peak appears at Time 0 ms.

Then, Xcorr between the second 1s waveform and the 2s waveform
[image]
However, the peak appears at 1009ms (not 1000ms if there is only phase difference)
 
It seems that there is not only phase difference but also time difference. I did some more tests and found the time error are not always the same value. (sometimes minus and sometimes plus, but the peak appears around the expected time lag)
 
Could someone fix this bug or give some more clues? Thanks!

osqzss, 2016-09-07:
Would you try to generate the I/Q samples without the carrier and see how things go?

`//ip = chan[i].dataBit * chan[i].codeCA * cosTable512[iTable] * gain[i];`
`//qp = chan[i].dataBit * chan[i].codeCA * sinTable512[iTable] * gain[i];`
`ip = chan[i].dataBit * chan[i].codeCA * 100;`
`qp = chan[i].dataBit * chan[i].codeCA * 100;`

CollusW, 2016-09-09:
After I do this change, the generated waveform samples are all zeros. :( anything missed?

osqzss, 2016-09-09:
Sorry I forgot the carrier amplitude. 

`ip = chan[i].dataBit * chan[i].codeCA * 250 * 100;`
`qp = chan[i].dataBit * chan[i].codeCA * 250 * 100;`

CollusW, 2016-09-09:
Good news. the samples are consistent with the new equation!
the combination of two short waveform files matches the long one. (differences are all zeros)
[image]

also the correlatioin are correct.
[image]
peak locates exactly point 1023000 which corresponds to 1s.

osqzss, 2016-09-09:
Okay, then there is no time (or code phase) difference between them.
The carrier phase is the only parameter to cause that waveform mismatch.
Because the carrier phase is initialized at the start time, the resulting waveform 
is also sensitive to it.

CollusW, 2016-09-09:
one more question. Is this carrier corresponds to the doppler shift? I am a little confused. I assume that we generate the baseband signal from this project and up-convert the baseband to RF using some SDR HW. If so, we should not care about the carrier phase at baseband. Could you pls explain what does the "carrier" mean?

osqzss, 2016-09-09:
It is the intermediate frequency (IF) wave with 0Hz center frequency and Doppler.

CollusW, 2016-09-09:
OK. I see. Thanks for your clearification.
Then, how do we solve the initial phase issue for the project? return the end phase and pass a parameter that specify the phase in the command line? any other better ways?

closed 2016-11-10 by osqzss