osqzss/gps-sdr-sim #50: Problem simulating when using USRP B200

andrmarkv, 2016-10-15:
Hello Takuji,

I am trying to use pgs-sdr-sim using USRP B200. I was able to compile your code, generate gpssim.bin file for static location and tried to transmit it using both gps-sdr-sim-uhd.py and tx_samples_from_file. When I start transmission process my B200 shows red light next to RX/TX antenna so I assume it does transmit something. I do not get any error and only UHD thread priority warning message during all my steps but when I use AndroidTS GPS Test or another android app called GPS Test by Chartcross Limited both of these apps shows me zero satellites (see screen shots). I am very new to SDR – that is my first real SDR project.

I was trying to follow instructions from your README.md file and then all advices and commands discussed in the ticket 31 (https://github.com/osqzss/gps-sdr-sim/issues/31) but still no success – my phone (actually couple of phones) do not show any satellites.

[image]
[image]

In addition to your advice I would like to get your help with several more questions that might be helpful and relevant to my situation:

1. Is is mandatory to use attenuator?
2. Does it really matter what antenna to use if my B200 and phone are 1 m from each other?
3. Generically speaking what is the troubleshooting process for that type of issues? Is it possible to use same USRP device to transmit and receive so I can at least see that I am transmitting something?
4. Does it really matter what brdc file to use during generation of the gpssim.bin file?
5. Do I have to use different brdc files for different geographical area (for example can I use same file for North America and Europe) during real simulation?

Below are commands and their outputs that might give you clues into what I am doing wrong

1. Generation of the gpssim.bin file:

```
./gps-sdr-sim -v -e brdc0420.16n -b 16 -s 2461538 -l 1.362334,103.992769,100
Using static location mode.
     1.583e-08   -7.451e-09   -5.960e-08    5.960e-08
     1.229e+05   -6.554e+04   -1.311e+05   -6.554e+04
     3.72529029846e-09   1.42108547152e-14     589824      1883
    17
Start time = 2016/02/11,00:00:00 (1883:345600)
Duration = 300.0 [sec]
02   65.2  36.0  22208138.2   4.9
05   20.0  18.7  23768071.9   6.7
12  207.1  29.2  23017076.1   5.2
13   39.9  58.7  20900389.1   3.4
18  269.8  10.7  24775535.2   6.4
19  145.6   6.4  25367843.4  10.1
20  326.8  30.7  22614357.9   4.8
24  181.8  26.8  23065651.5   5.7
25  248.3  22.3  23610114.8   5.5
29  324.2  32.7  22593974.7   4.6
32   34.2   3.1  25710384.1  10.2
Time into run = 30.0

…

Time into run = 270.0
02   67.9  36.3  22173139.1   5.0
05   21.6  17.8  23856379.3   7.0
12  205.0  28.3  23089866.9   5.4
13   37.6  56.8  20987167.9   3.5
18  271.8  11.0  24764817.0   6.5
19  146.0   4.8  25536818.2  10.7
20  329.0  30.2  22649745.7   5.0
24  180.2  28.0  22954061.2   5.7
25  246.1  22.0  23635617.4   5.7
29  323.1  34.5  22447713.4   4.6
32   34.1   1.5  25886836.8  10.8
Time into run = 300.0
Done!
Process time = 54.5 [sec]
```

2. Transmission of the generated data

2.1 using  tx_samples_from_file program:

```
/usr/lib/uhd/examples/tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0 --repeat
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam

Creating the usrp device with: master_clock_rate=50e6...
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 50.000000 MHz... 
-- Actually got clock rate 50.000000 MHz.
-- Performing timer loopback test... pass
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2

Setting TX Rate: 2.500000 Msps...
Actual TX Rate: 2.500000 Msps...

Setting TX Freq: 1575.420000 MHz...
Actual TX Freq: 1575.420000 MHz...

Setting TX Gain: 0.000000 dB...
Actual TX Gain: 0.000000 dB...

Checking TX: LO: locked ...
Press Ctrl + C to stop streaming...
^C
Done!
```

2.2 using  gps-sdr-sim-uhd.py  script:

```
python gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 40.000000 MHz... 
-- Actually got clock rate 40.000000 MHz.
-- Performing timer loopback test... pass
Using Volk machine: avx_64_mmx_orc
Press Enter to quit:
```

Thank you
Andrey

osqzss, 2016-10-15:
Please try the following two commands that HughAndersonNUS managed to sample at 2.5MHz in #31:

./gps-sdr-sim -v -e brdc0420.16n -b 16 -s 2500000 -l 1.362334,103.992769,100
./tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0 --repeat

osqzss, 2016-10-16:
Regarding your questions:

1. Is is mandatory to use attenuator?
It is not mandatory, but even the minimum TX power of the B200 is VERY strong. You could easily confuse the smartphone and car navigation systems of your of your neighbours.

2. Does it really matter what antenna to use if my B200 and phone are 1 m from each other?
No. It depends the efficiency of the antenna at GPS band.

3. Generically speaking what is the troubleshooting process for that type of issues? Is it possible to use same USRP device to transmit and receive so I can at least see that I am transmitting something?
You can use GNSS-SDR software receiver to record and analyze the simulated signals.
https://github.com/gnss-sdr/gnss-sdr

4. Does it really matter what brdc file to use during generation of the gpssim.bin file?
The RINEX version has to be 2.x. 

5. Do I have to use different brdc files for different geographical area (for example can I use same file for North America and Europe) during real simulation?
No. the brdc is a merge of the individual site navigation files into one.

andrmarkv, 2016-10-16:
Hello Takuji,

Thank you for your quick response and answers. 

I tried suggested commands:

andrmarkv, 2016-10-16:
Hello Takuji,

Thank you very much for  your quick response and answers.

I tried suggested commands:

```
/gps-sdr-sim -v -e brdc0420.16n -b 16 -s 2500000 -l 1.362334,103.992769,100

./tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 0 --repeat

linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam

Creating the usrp device with: master_clock_rate=50e6...
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 50.000000 MHz... 
-- Actually got clock rate 50.000000 MHz.
-- Performing timer loopback test... pass
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B200
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2

Setting TX Rate: 2.500000 Msps...
Actual TX Rate: 2.500000 Msps...

Setting TX Freq: 1575.420000 MHz...
Actual TX Freq: 1575.420000 MHz...

Setting TX Gain: 0.000000 dB...
Actual TX Gain: 0.000000 dB...

Checking TX: LO: locked ...
Press Ctrl + C to stop streaming...
```

Again, I can see red light on B200 but phone (actually I am testing with several phones) show no satellite at all.

In your answers option 3 (GNSS-SDR) sounds interesting, but I have same fundamental question that goes with it. Can I use same USRP device to transmit and receive at the same time. 
To test it I built a simple Flow Graph in GRC using only two blocks: USRP source and waterfall sink. I can run that flow graph only when I am not replaying stream using tx_samples_from_file utility. Each one of them (my flow graph or tx_samples_from_file) runs separately but if I, for example, try to run flow graph and at the same time start tx_samples_from_file it shows me following error:

```
./tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --typeshort --rate 2500000 --freq 1575420000 --gain 0 --repeat
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam

Creating the usrp device with: master_clock_rate=50e6...
Error: LookupError: KeyError: No devices found for ----->
Device Address:
    master_clock_rate: 50e6
```

That is why I am afraid that option 3 (GNSS-SDR) is not going to be helpful in my situation.

What else can you suggest to help me with troubleshooting.

Thanks
Andrey

osqzss, 2016-10-17:
Sorry I'm not a USRP user and unable to answer your USRP/GRC related question.

GNSS-SDR should work with the RTL-SDR dongle. 
http://www.rtl-sdr.com/buy-rtl-sdr-dvb-t-dongles/
It's cheap and very handy.

andrmarkv, 2016-10-29:
Hello Takuji,

I finally managed to make it all work. There were several problems that were introducing difficulties:

1. Problem number one was the fact that I had to transmit with high gain value. In all examples that I saw transmit gain was 0 or a value less then 50. I am not sure if it is something specific to my B200 or to all of them but I was able to detect ANY signal generated by my board only if I used gains above 70. So as soon as I changed parameter --gain 75 it started to work.

2. Second problem was in the script gps-sdr-sim-uhd.py. I think something is not right there. When I use command

`tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 80 --repeat`

it works but when I use

`./gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 80`

it ether does not work at all or I am getting only one, two green bars in the GPS Test android app. Rest of the bars are grey and phone never locks to GPS. My guess is that there is something wrong with clock rate as tx_samples srcipt shows on the screen

```
-- Asking for clock rate 50.000000 MHz... 
-- Actually got clock rate 50.000000 MHz.
```

but gps-sdr-sim-uhd.py script shows

```
-- Asking for clock rate 40.000000 MHz... 
-- Actually got clock rate 40.000000 MHz.
```

3. One more problem was in the fact that you have to use gps-sdr-sim tool with proper  -b and -s options

`./gps-sdr-sim -v -e brdc0420.16n  -b 16 -s 2500000 -l 1.362334,103.992769,10`

when I was missing them during generation of the gpssim.bin situation was similar to problem 2 - all lines in the GPS Test where grey and phone was not locking to GPS.

Having said all that I managed to use USRP b200 now to simulate fake location using commands:

```
./gps-sdr-sim -v -e brdc3020.16n -b 16 -s 2500000 -l 1.362334,103.992769,100
tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 80 --repeat
```

Thank you for a great tool
Andrey

osqzss, 2016-10-31:
Thanks for the tip.

I understand that you need to specify the sampling frequency with "-s" option,
but it is a bit unclear to me why you need "-b".

The sampling data format is 16-bit I/Q as default, and the code should generate 
the same binary file without "-b" option.

Are you sure that you need both the "-b" and "-s" options to make it work?

bitdust, 2016-10-31:
@osqzss

I test the code on my B210. 
It works fine without “-b”.

andrmarkv, 2016-10-31:
Confirming it works without -b option.

Takuji, one more question if you don't mind before resolving that issue. Do you know why during simulation phone, according to the GPS Test application, is loosing 3D GPS fix time to time. Application still might show a lot of green bars but it looses GPS for 10-15 seconds. Then, usually, it recovers back.

Could it be related somehow to the fact that I am replaying same data stream over and over and it is somehow detected by the phone, for example trough value of the time extracted from the GPS signals?

I am attaching several screenshots captured during simulation. They show normal status with 3D fix, then situation when I have 2D fix and finally no fix at all. All these screens where captured within single 5 minutes interval.

Once again
Thank you

[image]

[image]

[image]

osqzss, 2016-10-31:
I haven't seen any intermittent loss of navigation solutions with my bladeRF and ASUS ZenFone 2 setup. It is hard to tell what exactly causes your issue.

In your first picture, the signal strength looks still too high. Please add more attenuation and keep it somewhere around 50dBHz or lower.

andrmarkv, 2016-11-10:
Hello Takuji,

I decreased gain value of the tx_samples_from_file tool (I made it 40) which resulted in the levels of the sattelite signals going down to about 30-40. I do not have any attenuators so all my configuration is based only on the software settings.

I also did much more testing and it looks like there are two reasons why GPS Test app shows loss of locaton:

1. If you put application on the background and then bring it forward - it usually takes about 5-10 seconds to get a fix. Sometimes even faster.

2. When gps signal replay goes into the loop, so at the end of 300 seconds (5 miunutes) interval it looses signal for about 20-30 seconds but I believe that is expected behavior.

One more time thank you for you software and i believe you can close my ticket as resolved

Andrey

osqzss, 2016-11-10:
Hi Andrey,

Glad to hear that it is now working smoothly. 
Enjoy happy hacking!

closed 2016-11-10 by osqzss

devj27, 2020-11-10:
Hi,

I am able to generate GPS Signal using 
below commands

./gps-sdr-sim -v -e brdc3020.16n -b 16 -s 2500000 -l 1.362334,103.992769,100
tx_samples_from_file --args="master_clock_rate=50e6" --file gpssim.bin --type short --rate 2500000 --freq 1575420000 --gain 80 --repeat

My requirement is to generate Gps Signal  for Fix on daily basis  with that date 
i could able to achieve it by arg -T now while generating gpsbin.

problem what i am facing is the fix is not consistent  some time its getting fixed but some time its not.So how could i make it consistent to get continuous gps fix,

Devj

james-999, 2020-11-26:
this is an old topic. when i search in internet about how to use B210 simulate gps signal the most likely useful content is this topic but after i try as it suggested in this topic there are still many cellphone that can not detect the simulated gps signal. finally i find this [https://github.com/kobekobe1234567/gps-sim-ui](url) which is a windows ui programme that works with b210 however it need a external 10Mhz clock. i use Chinese produced B210 and a 0.01PPM 10MHZ OCXO it works very well for any cellphone or GPS module the location is very accurate. for some cell phone you need to uncheck the "use current time" check box otherwise the cellphone can not be spoof.