osqzss/gps-sdr-sim #52: Not working on hardware with ADRF6755

Vakaras-x, 2016-11-05:
Hello,

    I'm trying to simulate GPS in real time, (without creating files). All "time-machine" issues I solved already, 260k samples I generate in 50mS only! But GPS receiver (android phone with AndroiTS GPS Test) does not understand it. 
My hardware is based on ADRF6755 I/Q modulator. 
    I compared my generated digits with gpssim.bin. Content is identical. I used the last brdc3090.16n RINEX file. Reference clock is 2.5ppm
  Tryed to use various attenuators upto 47dB (integrated in ADRF6755). 
Here is a spectrum, connected directly to the output with 40dB attenuator:
https://drive.google.com/file/d/0B-O8KdbrAqdCNVVwVk5Bbk9OYXM/
https://drive.google.com/file/d/0B-O8KdbrAqdCOVNxSEwwY1RqUXc/

The location is fixed (option-l in gps-sdr-sim). 
I update my DAC at 2.6MHz sample rate. What could be wrong?

Thanks in advance for any comments.

IvanKor, 2016-11-05:
1. Разместить android phone with AndroiTS GPS Test так что бы он не пинимал реальных спутников.
2. При небольшой выходной мощности 40dB attenuator не обязателен. Я вообще ими не пользуюсь.
3. Проверить что windows обеспечивает слитную передачу без пропусков. Например мой i7 под win7 может по Tx USB может прекращать передачу аж до 5 милисекунд.
4. И т.д. например, проверить правильность полярности на Tx "0" и "1" в ADRF6755, возможно она инверсная.

Vakaras-x, 2016-11-05:
There is no Windows in the system. Just Cortex-A5, FPGA+DAC+ADRF6755
I put the scope on "start packet" pin, and always see nice pulse every 100mS. 
I know the possibility, that I and Q signals may be switched. So I made it switchable from user interface. But no difference so far.

IvanKor, 2016-11-05:
Тогда используя типа https://github.com/taroz/GNSS-SDRLIB проверить излучается ли правильно хотя бы преамбула. Если преамбула излучается то искать ошибку  в SW Cortex-A5.

Vakaras-x, 2016-11-05:
Спасибо. Ценный намёк. Но для воспользования - неделя роботы (финальные данные у меня в FPGA). Если не найду других вариантов - начну идти по этом пути.

closed 2016-11-18 by osqzss