osqzss/gps-sdr-sim #56: Add use nav file from RTKLIB

IvanKor, 2016-12-16:
Dear OSQZSS !
Many thanks to you for hard work.

Is it possible to add the option to gpssim use raw-data ublox instead BRDC (brdcXXXX.XXn)
or instead use BRDC add use nav file from RTKLIB ?

RTKLIB can make from "raw-data" ublox 6,7,8 RINEX files like BRDC.

osqzss, 2016-12-16:
gps-sdr-sim can read RINEX 2.10 navigation file. It doesn't have to be BRDC.

IvanKor, 2016-12-17:
Thanks very much !
RINEX 2.10 works.
My question appeared as I began to try RINEX 2.12 and UP. They did not work.

I modified your real-time signal generation bladeGPS to hackrfGPS.
Windows 7, i7-4710HQ @ 2.50GHz CPU load ~ 6%   
If you want I can send you the source code hackrfGPS.

osqzss, 2016-12-18:
That sounds nice. I am happy to test the code.

IvanKor, 2016-12-18:
I sent the source code.

closed 2017-01-20 by osqzss