osqzss/gps-sdr-sim #58: Subframe 1 maybe bug

IvanKor, 2017-01-28:
unsigned long ura = 0UL;
	//unsigned long ura = 2UL;//!!!

osqzss, 2017-01-28:
You're right. URA index 0 is more appropriate.

IvanKor, 2017-02-01:
Ok! Next ...

	unsigned long code = 1UL;
	unsigned long svh  = 0UL;

//	sbf[0][2] = ((wn & 0x3FFUL) << 20) | (ura << 14) | (((iodc >> 8) & 0x3UL) << 6);//!!!
	sbf[0][2] = ((wn & 0x3FFUL) << 20) | ((code & 0x3UL) << 18) | ((ura & 0xFUL) << 14) | ((svh & 0x3FUL) << 8) | (((iodc >> 8) & 0x3UL) << 6);

osqzss, 2017-02-02:
Fixed.

CollusW, 2017-02-04:
Since previous code still works as I tested, I wonder what is the phenomenon if it is not fixed? Thanks! @IvanKor

IvanKor, 2017-02-04:
Yes, gps-sdr-sim works fine, thanks Takuji!

However ... to play back the  nav-file  need fixes.
For example, nav-file has :
 4 17  2  2  8 44 27.2 -.260358676314E-03  .341060513165E-12  .000000000000E+00
     .200000000000E+02 -.227500000000E+02  .517164399116E-08  .595279177481E+00
    -.118464231491E-05  .879350677133E-02  .101700425148E-04  .515372814751E+04
     .381600000000E+06 -.502914190292E-07 -.246273129158E+01  .856816768646E-07
     .982721336208E+00  .195281250000E+03  .871349078547E+00 -.795354558281E-08
    -.281440294547E-09  .100000000000E+01  .193400000000E+04  .000000000000E+00
     .200000000000E+01  .630000000000E+02 -.195577740669E-07  .200000000000E+02
     .377067218000E+06  .400000000000E+01
svh  = 0x3F
SV4 OFF.
Without fixes : svh  = 0, SV4 ON.

osqzss, 2017-02-10:
It is not really a bug. You can now simulate even unhealthy satellites.

closed 2017-02-10 by osqzss

CollusW, 2017-02-10:
thanks for your explanations.