osqzss/gps-sdr-sim #60: A couple of doubts about the code

amc92, 2017-02-09:
Hello,
I am looking at these great codes, and I have found some trouble trying to understand a couple of things. I would be very grateful if someone could give me a hand with these topics, which are:

1.- I dont understand why, to calculate phase_ini, this expression is used: "phase_ini = (2.0*r_ref - r_xyz)/LAMBDA_L1;". I don't understand the part "2.0*r_ref", being r_ref the range between the satellite and (0,0,0) position (I think).

2.- In computeCodePhase, in the expression "ms = (((chan->rho0.g.sec-chan->g0.sec)+6.0) - chan->rho0.range/SPEED_OF_LIGHT)*1000.0;", I don't understand why 6 seconds are added. I guess it is necessary to skip the first 10 words, but I would like to know what exactly those words are, and what they are useful for (I have seen that dwrd has 60 elements, and I would like to understand why it doesn't have only the 50 words contained in 5 subframes).

Thank you very much

IvanKor, 2017-02-09:
To specify the exact time at all you need to add 18.0 seconds.
Interestingly, some GPS-receivers from true satellites give the exact time + 18 sec :)

osqzss, 2017-02-10:
amc92,

1. It is still experimental, and I haven't yet found a proper way to initialize the carrier phase.
2. The Z-count in the HOW word corresponds to the time of signal transmission of the "next" subframe. So you need to add 6 seconds (the length of each subframe) to specify the current code phase.

The 18-sec offset is the GPS-UTC time difference after the leap second on the 1st of January, 2017.

amc92, 2017-02-10:
Thank you both very much for your answers!

ttlewi, 2017-02-10:
Just a note on point 1, the Navstar system makes no guarantees about the
position of carrier phase relative to code phase and one can see that in
practice it is random across the constellation.  I rather like the way the
carrier phase is initialized now, it appears random between satellites and
simulations, but will produce the same initial phase value for a given
satellite between simulations with identical parameters.

closed 2017-02-11 by osqzss

CollusW, 2017-10-14:
What if we just initialize all the carrier phases to 0 degree? 
Could that be a problem to receivers?

osqzss, 2017-10-14:
If you are interested in only pseudorange-based single positioning, you can initialize all the carrier phase to 0 degree.

CollusW, 2017-10-14:
What kind of receivers can have problem for all 0 initial carrier phase? RTK? could you please explain a little? Thanks.

osqzss, 2017-10-14:
Any receiver can track the signal with the zero-degree initial carrier phase. But if you want to use the carrier phase measurements, you have to emulate the initial integer ambiguity of each carrier phase properly. Otherwise, the RTK software cannot resolve the ambiguity as an integer number. You cannot simply set it to zero or any fixed float number.

CollusW, 2017-10-14:
Thanks. What is the requirement of initial carrier phase for RTK software?

osqzss, 2017-10-14:
This article could give you a little insight of the carrier phase generation:
http://www.insidegnss.com/node/4368

CollusW, 2017-10-20:
I modified the code to initialize the phases to all zeros and did some tests. The result is as expected: receivers can fix the location and time correctly.

Regarding the phases for RTK, I read the above article and some other papers regarding RTK. I do not think RTK needs knowledge or requirements about the SV carrier initial phases, because it uses the phase difference between the receiver and baseline receiver (but not the initial phase).

I wonder if some one has done any RTK tests with this code?

osqzss, 2017-10-20:
https://github.com/osqzss/gps-sdr-sim/tree/master/rtk

CollusW, 2017-10-20:
Does this mean that the code can work with RTK?
what if carrier phase set to 0s?