osqzss/gps-sdr-sim #65: Some devices can not positioning

Xbalien, 2017-04-06:
Hello,
The project is useful for me.Thanks for your great job. I use this simulator to send gps signal by USRPB200.  But some devices are not effective.
I found that N5,Pixel,Car can get the spoofed position, but the HuaWei, XiaoMi,ViVo  are not effective. What is wrong here? Can you provide some advice to solve it or let me know where the problem is? Thanks.

bitdust, 2017-04-06:
It works on my HuaWei mate8.
you can change the position settings of your phone to use GPS only and open some app such as GPS tester to  accelerate the positioning.

Xbalien, 2017-04-06:
Thanks, I change the position settings of HuaWei mate7. It works for me. But the positioning time is too long and unstable. And other devices such as vivo X6S, Smartisan,XiaoMi MIX are not work. Whether multimode positioning will not be spoofed? such as WIFI,BEIDOU.

bitdust, 2017-04-06:
I have read some paper about WiFi-based position spoofing. set up some fake
APs by openwrt or something else.

closed 2017-04-12 by osqzss