osqzss/gps-sdr-sim #66: hackrf_start_?x() failed: USB error (-1000)

wdx7266, 2017-04-06:
Hello,
Thanks for your great job. My hackrf one meets some problem, how could I slove it?

```
sudo hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0 -R 
call hackrf_set_sample_rate(2600000 Hz/2.600 MHz)
hackrf_start_?x() failed: USB error (-1000)
Usage:
	-h # this help
	[-d serial_number] # Serial number of desired HackRF.
	-r <filename> # Receive data into file (use '-' for stdout).
	-t <filename> # Transmit data from file (use '-' for stdin).
	-w # Receive data into file with WAV header and automatic name.
	   # This is for SDR# compatibility and may not work with other software.
	[-f freq_hz] # Frequency in Hz [0MHz to 7250MHz].
	[-i if_freq_hz] # Intermediate Frequency (IF) in Hz [2150MHz to 2750MHz].
	[-o lo_freq_hz] # Front-end Local Oscillator (LO) frequency in Hz [84MHz to 5400MHz].
	[-m image_reject] # Image rejection filter selection, 0=bypass, 1=low pass, 2=high pass.
	[-a amp_enable] # RX/TX RF amplifier 1=Enable, 0=Disable.
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable.
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-x gain_db] # TX VGA (IF) gain, 0-47dB, 1dB steps
	[-s sample_rate_hz] # Sample rate in Hz (4/8/10/12.5/16/20MHz, default 10MHz).
	[-n num_samples] # Number of samples to transfer (default is unlimited).
	[-S buf_size] # Enable receive streaming with buffer size buf_size.
	[-c amplitude] # CW signal source mode, amplitude 0-127 (DC value to DAC).
	[-R] # Repeat TX mode (default is off) 
	[-b baseband_filter_bw_hz] # Set baseband filter bandwidth in Hz.
	Possible values: 1.75/2.5/3.5/5/5.5/6/7/8/9/10/12/14/15/20/24/28MHz, default <= 0.75 * sample_rate_hz.
	[-C ppm] # Set Internal crystal clock error in ppm.
	[-H hw_sync_enable] # Synchronise USB transfer using GPIO pins.
```

Beyond above, when I make it, something happened:

> gps-sdr-sim# make
> gcc -O3 -Wall   -c -o gpssim.o gpssim.c
> gpssim.c: In function ‘main’:
> gpssim.c:1860:65: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]
>   if (duration<0.0 || duration>((double)USER_MOTION_SIZE) / 10.0 && !staticLocationMode || duration>STATIC_MAX_DURATION && staticLocationMode)
>                                                                  ^
> gpssim.c:1860:120: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]
>   if (duration<0.0 || duration>((double)USER_MOTION_SIZE) / 10.0 && !staticLocationMode || duration>STATIC_MAX_DURATION && staticLocationMode)
>                                                                                                                         ^
> gcc gpssim.o -lm -o gps-sdr-sim
>

osqzss, 2017-04-07:
Please post your HackRF related question to their forum. You may get more help there.
Regarding the warnings, they should be fixed in the current version.

wdx7266, 2017-04-08:
Thank you very much!

closed 2017-04-10 by osqzss