osqzss/gps-sdr-sim #67: GLONASS support?

MAVProxyUser, 2017-04-06:
Can this code work with GLONASS if the proper ephemerides are used?

osqzss, 2017-04-07:
No. It currently supports GPS only.

MAVProxyUser, 2017-04-07:
Do you know what semantics are involved in adding GLONASS support? Could you suggest which code paths would need to be changed / altered or branched into?

osqzss, 2017-04-07:
GPS is using CDMA while GLONASS employs FDMA. It requires much wider bandwidth thus higher sampling rate.

MAVProxyUser, 2017-04-09:
So based on what we know about the signal, do you think it could be integrated into your codebase, or would it need to be a separate project altogether? 
https://www.naic.edu/~phil/rfi/sat/glonass/GLONASSOverview.pdf

[image]

"Unlike GPS, all GLONASS satellites transmit the same code at different frequencies. They derive signal timing and frequencies from one of three on-board cesium atomic clocks operating at 5 MHz:
For example,
L1 = 1602 MHz + (n x 0.5625) MHz where
n = the frequency channel number (n = 0, 1, 2 and so on)
It means that satellites transmit signals on their own frequency, separated by multiples of 0.5625 MHz
or 562.5 kHz, from the frequency of other satellites; see Figure 2 below."

osqzss, 2017-04-09:
The center frequency of GLONASS signal is different from that of GPS, and most SDR devices support only one TX channel. There is no strong reason to integrate GLONASS code into gps-sdr-sim. It can be a separate project.

MAVProxyUser, 2017-04-09:
"no strong reason to integrate GLONASS code" Understood... the center frequency could be an option on command line. For what it is worth, as far as multi channel transmission... please see the above example where only a single channel was used successsfully. 

'' tx_samples_from_file --type short --repeat --spb 13500000000 --rate 8333333 --freq 1602000000 --gain 15''

osqzss, 2017-04-09:
Yes, all the visible GLONASS signals can be transmitted from a single TX port with about 12MHz bandwith. But its lowest frequency is still >20MHz higher than the GPS L1 band.

MAVProxyUser, 2017-04-09:
for the sake of conversation... would it be obtuse to fork the project and add an option to keep the existing work flow and use a command line flag to specific GLONASS? I am looking at the raw semantics of it. What you have now *works* no need to change it drastically if we can simply add a module for example.

osqzss, 2017-04-09:
gps-sde-sim is not designed for other GNSS signals. I think it would be more effective to create a new project with a general architecture and design rules for all GNSS signals.

MAVProxyUser, 2017-04-09:
Thanks for your input!

closed 2017-04-10 by osqzss

MAVProxyUser, 2017-04-30:
FWIW... looks like FDMA glonass is considered "legacy"... http://www.novatel.com/assets/Documents/Downloads/NovAtelChartH.pdf

The newer L3 signal is indeed CDMA.

gasparjoao, 2018-05-14:
Hello guys
I've been reading your entire conversation and I also find it very interesting to be able to simulate GLONASS signals.
osqzs and MAVProxuUser can you tell me some informations about applying GLONASS data to GPS-SDR-SIM that makes it possible to transmit these signals?

Many thanks for your availability, best regards

osqzss, 2018-05-14:
The architecture of gps-sdr-sim is not very adaptable to GLONASS system. It is possible to generate GLONASS signals, but you need rewrite almost all the code.

gasparjoao, 2018-05-14:
Oh okay and do you know any kind of system that can generate GLONASS signals?

osqzss, 2018-05-14:
There are many commercial simulators capable of generating GLONASS signals but as far as I know no open source one.

ALDANIZ, 2022-02-04:
MAVProxyUser
Have you developed the gps-sdr-sim software for glonass?

Retinaaa, 2023-07-02:
@MAVProxyUser @osqzss
do you know any kind of system that can generate GLONASS signals?

Gerald0217, 2024-01-09:
> FWIW... looks like FDMA glonass is considered "legacy"... http://www.novatel.com/assets/Documents/Downloads/NovAtelChartH.pdf
> 
> The newer L3 signal is indeed CDMA.

Dear expert,

The specific query I have is: How to generate GLONASS signals? I am interested in understanding the technical aspects and processes involved in generating signals for GLONASS navigation.

If you could spare some time to share your expertise on this matter, I would greatly appreciate it. Any resources, references, or guidance you can provide will be immensely helpful.

Thank you very much for considering my inquiry. I look forward to hearing from you.

Best regards