osqzss/gps-sdr-sim #69: Can we merge in the FrankBuss STDIN fixes sans the nodeJS?

MAVProxyUser, 2017-04-11:
Can we just make this a standard feature? https://github.com/FrankBuss/gps-sdr-sim

We can leave out @frankbuss JavaScript stuff in the maps folder, then anyone wanting to use it can make use of what ever front end they chose to. 

https://github.com/FrankBuss/gps-sdr-sim/tree/master/maps

I wanted to get his fixed put in place for the *current* version of code, but am not really interested in doing that for every chance you make. 

Can we get some IF statements, and command line statements to accommodate his modifications? 

THIS https://github.com/FrankBuss/gps-sdr-sim is way more practical than THIS for example... 

https://www.douban.com/note/572829951/

This guy completely hijacks all of fopen as a fix... https://www.douban.com/note/572588515/

Surely we can just accommodate folks that use this functionality over STDIN?

osqzss, 2017-04-11:
gps-sdr-sim is not designed for real-time simulation, and I am not very keen to merge his STDIN feature to the code. You can find some more discussion in issue #41.

MAVProxyUser, 2017-04-12:
I understand that you haven't designed it for that, but clearly people are having success with it and doing it anyway. I am suggesting you just add the few lines of code and be done with it. Many folks would happy. 

The #41 discussion first you said "because the named pipe is not available in Windows"... which was shown to be incorrect. Then when it was suggested it be rewritten as a library you said it was beyond your skill as a programmer. 

"You cannot simply select any location for the next time step. It would generate unrealistic user dynamics, and the receiver could never catch the signal. You need a path generator, but it is not trivial." I think some of us would be happy to work on a path generator with some constraints IF STDIN as a feature was part of the mainline code. 

Just something to think about... you created a monster, people are using it it ways you did not expect, *successfully*.

MAVProxyUser, 2017-04-12:
I believe this should be the @frankbuss patch moved to the current version of *master*. 
[frankbuss_stdin_patch_diff.txt](https://github.com/osqzss/gps-sdr-sim/files/915166/frankbuss_stdin_patch_diff.txt)

osqzss, 2017-04-12:
Do you know any way to make "unistd.h" and "fcntl.h" in the patch work on Visual Studio?

MAVProxyUser, 2017-04-12:
Use this for the unistd.h replacement for Windows... 
http://stackoverflow.com/posts/826027/revisions

This seems to cover the fcntl.h variation 
https://msdn.microsoft.com/en-us/library/z0kc8e3z.aspx

MAVProxyUser, 2017-04-12:
You could alternately support MacOS and Linux users with some IFDEFs and not even present the command line option to Windows until someone can make it work. I'll see if I can find a windows box to try sometime this next few weeks if you can't figure it out. 

#ifdef  defined(_WIN32) || defined(WIN32)   
...
// Do windows specific variation here. 
...
#endif

bitdust, 2017-04-12:
The current version is cool, just use the pure C to achieve the GPS signal generation. There are many kinds of SDR/OS/platform, everyone can easily understand the code and use it as the core of their system.
I finished my work on real-time simulation. I found it's hard to make it real time and keep the code still simple and clear. finally, you will have to use something like thread/mutex/queue to make it work perfectly, STDIN/STDOUT is far not enough.

MAVProxyUser, 2017-04-12:
@Aaron-Luo has done something similar to what you mentioned @bitdust. The problem is his variant is windows only. It needs reworked for Linux / OSX. 

https://github.com/Aaron-Luo/DEFCON24/blob/master/Realtime_generate_fake_GPS_by_joystick/src/hackrf_gpsspoof.cpp

bitdust, 2017-04-12:
@MAVProxyUser 
my work is based on CPP/BOOST and works on USRP device. 
The problem is everyone wants to build their own system on their devices. Just focus on the signal generation is enough and anyone else can do their own port work.

MAVProxyUser, 2017-04-12:
I am happy to use USRP... is your work shared anywhere?

bitdust, 2017-04-12:
@MAVProxyUser 
OK, I will share my code in Github tomorrow.

IvanKor, 2017-04-12:
Not only generating real-time GPS signals.
Necessary to generate ephemeris satellites that can not be received by the GPS receiver.
аnd generate ephemeris satellites that will appear over the horizon in the future.

osqzss, 2017-04-12:
If you have a BladeRF, try BladeGPS for real-time simulation. It should work on Linux.
https://github.com/osqzss/bladeGPS

MAVProxyUser, 2017-04-12:
Unfortunately generally don't use windows, likewise don't have a bladeRF.

closed 2017-04-23 by osqzss