osqzss/gps-sdr-sim #72: Something problem about  spoof signals

wafppp, 2017-05-07:
This project is very nice, it is useful for me. Now, I have some problem,  any people can help me?
1. I find that the subframe4 and  subframe5 are missing about almanacs. Has any plan to add it to the gps signals? how to download it?
2. I look up the code, but I don't catch the mainly idea of how to generate spoof signals by given receiver location.
Thanks.

osqzss, 2017-05-10:
I didn't implement the GPS almanac frames to the simulator since they are not required to get position fix.
You can download the GPS YUMA almanac files from USCG or CelesTrak:
https://www.navcen.uscg.gov/?pageName=gpsAlmanacs
https://celestrak.com/GPS/almanac/Yuma/
Adding these parameters into the subframes 4 and 5 is rather straightforward, and any contribution is always appreciated.

wafppp, 2017-05-12:
> I didn't implement the GPS almanac frames to the simulator since they are not required to get position fix.
> You can download the GPS YUMA almanac files from USCG or CelesTrak: [...]

Thanks. I will try it.
I found some case can't fix the position during my test.I try to send the signals to the phone which use the mode of high accuracy.I do not know what the problem is, could you tell me what is wrong?

osqzss, 2017-05-12:
It would be better to use the "GPS only" mode. In the "high accuracy" location mode, Wi-Fi and/or cellular based location is used apart from GPS.

IvanKor, 2017-06-03:
almanac frames is not simple, for example :
		Ch	SV	GNSS	FL	TOW/TTAG	MSG	DATA 
		0	 25	G25	X-	566160	5/22	8B0E7C  B84C34  563DB8  0FF376  FD2800  A10DF6  BD0075  B9CBE9  1B9295  FEFFB3 	   
		1	  4	G4	X-	566160	5/22	8B0E7C  B84C34  563DB8  0FF376  FD2900  A10DF5  BD0075  B9CBE9  1B9295  FEFFB2 	  
		3	 12	G12	X-	566160	5/22	8B0E7C  B84C34  563DB6  24F373  FD2400  A10DEE  BCF8C5  B9D051  1AB235  FEFFAA 	   
		9	  5	G5	X-	566160	5/22	8B0E7C  B84C34  563DB6  24F373  FD2400  A10DEE  BCF8C5  B9D051  1AB235  FEFFAA 	  
		14	 26	G26	X-	566160	5/22	8B0E7C  B84C34  563DB8  0FF376  FD2800  A10DF6  BD0075  B9CBE9  1B9295  FEFFB3 	  
		15	 29	G29	X-	566160	5/22	8B0E7C  B84C34  563DB6  24F373  FD2400  A10DEE  BCF8C5  B9D051  1AB235  FEFFAA 	  

		SV	WEEK										D0	    D1	    D2	    D3	    D4	    D5	    D6	    D7
		22	1952										563db8	0ff376	fd2800	a10df6	bd0075	b9cbe9	1b9295	feffb0

closed 2017-06-18 by osqzss