osqzss/gps-sdr-sim #73: Live spoofing question

Spoof4fun, 2017-05-11:
Hello, I am a research student in a US university and I have been tasked to create a live spoofer. I have been able to successfully run the code for a spoofed static location and predefined trajectory. My next scenario is receiving a live signal and manipulating it live. Much like the notable boating demonstration done by another research group which did it to a yacht. My scenario includes me being in a vehicle and my spoofing target being a few meters away. One of the problems I foresee is after the spoofing is activated I fear that I will spoof my own receiver. 

From my knowledge as it stand now I feel that I may be able to modify the code to work with a live stream of data but I am very short on time and don't have the liberty to explore hunches. 

My question is;
With modification can this code read a live stream of data and in what format?
I feel that is possible because an option in the code is to receive a NMEA GGA stream

I am very new to this area of research and lack a professor with expertise in the subject matter at my university so anyone with any knowledge or assistance they can offer is greatly appreciated.

Thank you

bitdust, 2017-05-11:
I think you can place your receiver far away from your spoofing area, and transmit the live NMEA information through Internet.
As for the stream-mode spoofing, you can check my real-time version: 
https://github.com/bitdust/gps-sdr-sim

Spoof4fun, 2017-05-11:
Thank you for your response bitdust.
I cannot place my receiver away from my spoofer because I will be in a moving car with my receiver while I try to spoof the car. 

Your real time version is very close to what I am looking for but instead of clicking on the map I am hoping to replace the clicking on the map with live position data received by my receiver and then altered slightly. 

I have considered the potential of receiving from the L2C band , altering the position for example by 5 meters to the left, and transmitting that location on L1C. 

Does the above sound like a reasonable approach? Thank you again for taking the time to advise me.

Spoof4fun, 2017-05-11:
Gym 487 seems to have created something similar to what I am looking for but he is using a TCP source running in server mode and sending data to an osmocon Sink which I assume I can change to a USRP sink.

I am currently trying to get in touch with him for more information about the TCP source and how it work. His work is here https://github.com/gym487/gps-sdr-sim-realtime . 

Would it be possible to stream real time gps data through that TCP source into a USRP sink?

miko87, 2017-05-18:
On the same topic of live gps spoofing. How can you make the spoofing start at a "future" time (the broadcast ephemerides only has data for the past, not the present / future)?
Is there a site that provides prediction for ephemerides files ?
I found a site that produces predictions for sp3 files but not bc format (RINEX).
Any clue?

Spoof4fun, 2017-05-18:
Well the theory I'm pursuing is using gnss sdr to connect to a live/true gps signal take that date,  manipulate it,  as in  change the data to show 5 meters to left and feed that into the gps sdr sim. Hopefully a continuous feed. I think it should work for the "present" scenario due to the live connection using the  gnss sdr although I expect a delay due to processing time.

miko87, 2017-05-21:
Was anyone actually able to use this prohject to conduct a live gps spoofing in an outdoor environment where GPS reception of real satellites is good?

Spoof4fun, 2017-05-21:
I'm working on achieving that this summer so I'll let you know by the end of it where I get and what was possible, but you won't be able to do it live with just this program because it doesn't have support to stream live on it's own. I plan in feeding it live data from gnss-sdr.

closed 2017-06-18 by osqzss

Honky103, 2017-08-28:
Hi Spoof4fun, I'm curious to learn about your progress in integrating gnss-sdr and gps-sdr-sim to do live streaming. What was your approach? I'm planning to do something similar and I'm thinking of using gnss-sdr to save the RINEX NAV data into a file, then running bladeGPS to read from it and stream it live. Do you think it'll be feasible?