osqzss/gps-sdr-sim #77: Segmentation fault

marcoD64, 2017-06-26:
Good afternoon,

Firstly thank you for your work.
When I try to execute your code with the option "-g triumphv3.txt" I can't and in the terminal it is written "Segmentation fault (core dumped)" do you know what is the problem?
To be clear with the option -l and -u that works but not with -g why?
Thank you

osqzss, 2017-06-26:
It works fine on my PC. Would you give us a little more details on your development environment?

[image]

marcoD64, 2017-06-26:
that's true it work on Wondows but not on Ubuntu 14.02

marcoD64, 2017-06-27:
the proof
(i modify only the data which are printed to be more clear for me)
[image]

osqzss, 2017-06-27:
What does the core file say?

marcoD64, 2017-06-27:
how can I see what the core file say?

osqzss, 2017-06-28:
I couldn't recreate the error on my Ubuntu 16.04 LTS with the current source files.
[image]

marcoD64, 2017-07-04:
thank you for your answer. I pass on Ubuntu 16 and download again all the files and it work. I don't know why but thank you.
I have also an other question concerning the signal strenght. What is the signal strenght of this programm? Or more preciselly what gain[i] represents?

osqzss, 2017-07-04:
gps-sdr-sim is a waveform generator and can't specify the TX power, which depends on your SDR device. The maximum amplitude of the waveform is about 2^10 peak-to-peak and nicely fits into 12-bit I/Q of BladeRF. If you set gain[i] to 100, it gives a full swing.

marcoD64, 2017-07-04:
ok thank you for your answer
and for hackRF One?

osqzss, 2017-07-04:
4-bit shift to the right.

marcoD64, 2017-07-04:
<span style="font-family:arial,helvetica,sans-serif; font-size:12px">i‌ thought it was 8 bit<br>
and this power in dB?</span>

<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : notifications@github.com<br>
A : gps-sdr-sim@noreply.github.com,marcolebasque@sfr.fr,state_change@noreply.github.com<br>
Envoyé: mercredi 5 juillet 2017 01:07<br>
Objet : Re: [osqzss/gps-sdr-sim] Segmentation fault (#77)<br>
 
<div class="gl_quoted">
<p>4-bit shift to the right.</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>

<div itemscope="" itemtype="http://schema.org/EmailMessage">
<div itemprop="action" itemscope="" itemtype="http://schema.org/ViewAction"> </div>
</div>
</div>
</div>
<link href="https://github.com/osqzss/gps-sdr-sim/issues/77#issuecomment-312965005" itemprop="url"> <meta itemprop="name" content="View Issue"> <meta itemprop="description" content="View this Issue on GitHub">
<div class="gl_quoted"> </div>

osqzss, 2017-07-04:
HackRF requires 8-bit I/Q. The default waveform is designed for 12-bit I/Q, and 4-bit shift to the right gives 8-bit I/Q waveform. Again, you can't specify the TX power with the gps-sdr-sim.

marcoD64, 2017-07-04:
oh thank you
I understand that I can't specify the TX power with gps-sdr-sim. But after the creation of gpssim.bin  is it possible to specify the power?

osqzss, 2017-07-04:
Use a spectrum analyzer to measure the TX power of your SDR device.

closed 2017-07-21 by osqzss