osqzss/gps-sdr-sim #80: transmitting

robellebor, 2017-07-31:
why am i getting "gps-sdr-sim-uhd.py: command not found" when i run the command "gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0" ????????

Honky103, 2017-07-31:
gps-sdr-sim-uhd.py is a python script. You should type "python gps-sdr-sim-uhd.py -t gpssim.bin -s 2500000 -x 0"  instead

robellebor, 2017-07-31:
thanks ... i fixed it by adding ./ before the .... but am having a problem transmitting it ... am not getting signal in my mobile phone .... i increased the gain in the gps-sdr-sim-uhd.py (def get_options()) ... but still nothing .... am using usrp x300 .... any idea on wats happening ??

Honky103, 2017-07-31:
Try disabling your assisted gps on your mobile. Then do it indoors or in a basement. Its easier to get a fix then. 

If it doesnt work, you have to check if your usrp is actually transmitting the signals. You can download apps from play store that allows you to see the signal strength. 

I use a usrp b210 with gps antenna with a gain of 20dB.

robellebor, 2017-07-31:
what kind of antenna r u using ???

sundw2014, 2017-07-31:
I think it doesn't matter what antenna you use. The  required GPS signal strength is really low, so all the antenna for X300 can be used to transmit GPS signal in this case. I tested this program with a 3G antenna and a log periodic antenna (850 MHz to 6.5 GHz). Both of them worked fine.
In my case, I turned off all the network connection of my phone and cleaned the A-GPS data.
You can use this android app to clean the A-GPS data and view the signal strength.
https://play.google.com/store/apps/details?id=com.pierwiastek.gpsdata
Enjoy^_^

Honky103, 2017-08-01:
I used TS.07.0113. But sundw2014 is right, you should be able to receive with the USRP antenna.

robellebor, 2017-08-09:
got it, thanks guys

Ramzeth, 2017-08-09:
I'm also using usrp x300 and have a trouble. When I replaying the samples some times i'ts all ok, my Android get the signal and successfuly fix on it, tested with [this](https://play.google.com/store/apps/details?id=com.chartcross.gpstest) app. But some times when receiving signal from usrp, i got a number a SV as it shoud to be, and when i got ephemeris data on that SV, the list of them is dropped. It seems like reset of GPS chip, which started to gather GPS data from beginning, but cannot fix position after that. Firsth thing i thought about my mobile device, but i tested the same case, same laptop, and the same time with bladeRF, and it works perfectly. All tests were performed in basement, so signals from real SV are absolutely absent. I activate a plane mode on my phone and disabled AGPS, so no any ephemeris data from other souces than gps-sdr-sim. At now i totally bewildered and cannot find a way even to start resolving this issue. Any ideas about this guys ?

Ramzeth, 2017-08-10:
Seems i've resolved the issue.
I've used tx_samples_from_file for replaying file with samples.
Samples should be generated with 16bit resolution.
`sudo /usr/lib/uhd/examples/tx_samples_from_file --file ~/tmp/gps_test_signal_16 --rate 2500000 --freq 1575420000 --gain 0 --type short `

closed 2017-08-15 by osqzss