osqzss/gps-sdr-sim #81: Documentation

robellebor, 2017-08-09:
Is there any further documentation on the gpssim.c file ..... am trying to understand the code and most of the declared variables have no description .... for example ...
double tk;
	double mk;
	double ek;
	double ekold;
	double ekdot;
	double cek,sek;
	double pk;
	double pkdot;
	double c2pk,s2pk;
	double uk;
	double ukdot;
	double cuk,suk;
	double ok;
	double sok,cok;
	double ik;
	double ikdot;
	double sik,cik;
	double rk;
	double rkdot;
	double xpk,ypk;
	double xpkdot,ypkdot;

Honky103, 2017-08-09:
It's documented in the IS-GPS-200 document.

robellebor, 2017-08-09:
ya i found it .... i was going to delete the issue ... u already answered it ... thnks

closed 2017-08-09 by osqzss