osqzss/gps-sdr-sim #82: gpssim.c

robellebor, 2017-08-15:
in the gpssim.c file there is a brief that says "Read ephemersi data from the RINEX navigation file" .... where is the RINEX navigation file .... or is it fetching it from the internet ???

osqzss, 2017-08-15:
You can find the RINEX navigation file of the daily GPS broadcast ephemeris (brdc) in the following archives:
ftp://cddis.gsfc.nasa.gov/gnss/data/daily/

robellebor, 2017-08-16:
so that means , we have to be online when we run the gps-sdr-sim ??

Honky103, 2017-08-16:
Nope there is a file named "brdc0xxx.14n" in the directory alr. It will read from the same directory as ur gps-sdr-sim

robellebor, 2017-08-16:
okay ... am just having a problem getting the gps signal in my phone "gps-test" mobile application .... i have tried by connecting the usrp to a spectrum analyzer and checked that the signal is being transmitted in the right frequency (1575.42 MHz) ,and it works fine,  but still i cant get the signal in my phone ???

Honky103, 2017-08-16:
Try in a gps-denied environment. Refer to issue #80. This was raised before

IvanKor, 2017-08-16:
RINEX navigation file at this time can always be obtained using a GPS receiver, for example,
u-blox --> now.nav 

[image]

Honky103, 2017-08-16:
Oh which ublox chip are you using?

IvanKor, 2017-08-16:
u-blox 6,7,8

Honky103, 2017-08-16:
Are you using the t-series or n-series

IvanKor, 2017-08-16:
Any,
u-blox 7.03 (45969) 00040007 --> NEO-6M-0-001
US $3.63
https://ru.aliexpress.com/item/1Set-GPS-Module-GY-NEO6MV2-NEO-6M-With-Flight-Control-EEPROM-RS232-TTL-Large-Antenna-For/32803522576.html?spm=a2g0v.search0104.3.41.sSiXHT&ws_ab_test=searchweb0_0,searchweb201602_1_10152_10170_10065_10151_10068_10084_10083_10080_10307_10082_10081_10110_10137_10111_10060_10112_10113_10155_10114_10154_10056_10055_10054_10311_10312_10313_10059_10314_10315_100031_10099_10078_10079_10103_10073_10102_10052_10053_10142_10107_10050_10125_10051,searchweb201603_1,ppcSwitch_7&btsid=fa3e5981-a2f6-4967-8ef3-fb93b141ae59&algo_expid=cc8cf34d-564a-476f-a40c-234b507ad924-5&algo_pvid=cc8cf34d-564a-476f-a40c-234b507ad924&transAbTest=ae803_2

Honky103, 2017-08-16:
Hmm i know this is a digression, but how do you get the RINEX navigation data file from the ublox chip?

IvanKor, 2017-08-16:
UBX-AID-EPH & UBX-AID-HUI
convert to now.nav 
UBX-AID-EPH-ALM
convert to almanac.sem.txt
Almanac is required for the synthesis of EPH satellites that could not take the GPS receiver.

Honky103, 2017-08-21:
Oh thanks IvanKor for the information. I am able to save UBX-AID-ALM to almanac.sem.txt from ucenter. However, I can't extract UBX-AID-EPH and UBX-AID-HUI to a file from ucenter. Any idea how do I do that? I am using ucenter_v8.26

IvanKor, 2017-08-21:
I do not use ucenter.
I use my software.

closed 2017-08-30 by osqzss