osqzss/gps-sdr-sim #83: Long duration testing

kiet, 2017-08-16:
Hi folks, I've been trying to run a test where I'm trying to set the duration to be at least a day. Usually around 4 to 8 hours into the test, I'm noticing satellites going out of visibility and eventually, no satellites visible at all.

Is there something I'm missing in order to run longer duration tests?

osqzss, 2017-08-16:
Each set of ephemeris data is only valid for about four hours. You need a RINEX navigation file which holds a long enough series of ephemerides.

kiet, 2017-08-16:
Hmm, I currently have a test running right now with yesterday’s daily file.

The test has the “-T now” option set. It’s been running for about 7.5 hours.

Do you have any recommendations on finding a longer RINEX file?

Thanks!

gpsdr, 2017-08-17:
hi kiet
I setup my test platform with usrp and AndroiTS GPS TEST software on my phone and used commands on gps-sdr-sim but I could not get any result. I can see the signal with spectrum analyzer but nothing on the test software. Any Idea.....??

osqzss, 2017-08-17:
It could be a bug in my code. Would you give me the exact command line and the RINEX navigation file that you used? By the way, did you run the gps-sdr-sim for such a long period? The file size is going to be gigantic!

kiet, 2017-08-18:
I'm using the latest gpssim that's been merged into bladeGPS. This 

is the RINEX file I'm using.
[brdc2280.zip](https://github.com/osqzss/gps-sdr-sim/files/1235398/brdc2280.zip)

osqzss, 2017-08-19:
Did you merge the latest gps-sdr-sim source into bladeGPS by yourself or simply use the latest bladeGPS code to run the long duration test?

kiet, 2017-08-19:
It's actually the latest bladeGPS code.

IvanKor, 2017-08-19:
brdc2280.17n --> 0:00 - 16:00 HH
16:00 HH - (T now) = 7.5 hours

kiet, 2017-08-19:
What would you recommend doing to lengthen the duration of the test to say, 3 days or longer? Are there tools to generate a longer RINEX nav file?

IvanKor, 2017-08-19:
Use almanac to synthesize inaccurate ephemeris (RINEX nav file).
Tools to generate a longer RINEX nav file does not exist. It's a secret (IODE ... e.t.s) .

closed 2017-08-30 by osqzss