osqzss/gps-sdr-sim #84: bug SC08

IvanKor, 2017-08-21:
If 12 satellites then the maximum value of i_acc or q_acc 255 x 12 = 3060
Then 3060 >> 7 = 24 and then again 24 >> 4 = 1
If you use hackrf with I / Q +/- 1 then this is a too rough approximation.
Hackrf like that works but with a very high level of noise.
Since the I / Q modulator levels are extremely small.
The normal levels are I / Q +/- 127.
Therefore, >> 7 for hackrf should be reduced to >> 5
And the next >> 4 to remove.

osqzss, 2017-08-21:
The first 7-bit shift is applied for the fixed-point arithmetic of the gain[] variable, which represents the free-space path loss and the receiver antenna gain. The value of 128 (= 2^7) means 0dB (= 1.0) of gain.

IvanKor, 2017-08-22:
Sorry, it's my mistake.
Now I'm converting Your sources into a DE0-nano-SoC FPGA and removing the gain.

osqzss, 2017-08-22:
So, are you making a stand-alone GPS simulator? Which RF front-end is on the top of the DE0-nano board?

IvanKor, 2017-08-22:
Yes, stand-alone GPS/GLONASS simulator.
In the first step I use hackrf via OTG USB ARM.
Coprocessor GPS/GLONASS on FPGA for ARM.
If hackrf via OTG USB ARM has problems then I'll think about what RF front-end to use.
If RF front-end then it will be necessary TXCO for DE0-nano board, etc.

closed 2017-08-30 by osqzss