osqzss/gps-sdr-sim #85: Access Denied- Windows 8

hunterlit64, 2017-08-30:
I got the sim on windows 8, but when trying to run 
" > gps-sdr-sim -e brdc3540.14n -l 30.286502,120.032669,100 "I get "access denied" and when running as an administrator it comes up it can't be found?

osqzss, 2017-08-30:
Make sure you run it from the command prompt. It doesn't work on the Windows PowerShell.

hunterlit64, 2017-08-30:
Sorry, how do you mean the command prompt?  I did it by typing "cmd" in the task manager and doing it from there.

osqzss, 2017-08-30:
It will open the command prompt. Did you run it from the folder where the gps-sdr-sim located?

hunterlit64, 2017-08-30:
Do you mean did I click on the exercise file? It shows up with the usage options such as  -e <gps_nav>  ,etc  in the cmd" I just can't run anything from it. I'm not used to Windows.

hunterlit64, 2017-08-30:
*exercise file* not exercise.

hunterlit64, 2017-08-30:
exe, hate spell correct sometimes:)

osqzss, 2017-09-06:
Type "cmd" in the address bar of the file explorer where gps-sdr-sim.exe is located.

closed 2017-09-14 by osqzss