osqzss/gps-sdr-sim #88: Exact condition for successful mission

EhsanArio, 2017-10-03:
Hi
I tested software many times with different situations. Some good and some bad results.
This is my setup :
1- Hackrf with a TCXO clock
2- a Telescopic Antenna
3- Laptop and ubuntu 16.04 running gps-sdr-sim
4- 10 Watts amplifier (in some tests)
NO attenuator is used

in some cases my phones locked and some not (GPS config set to "Device only" for all tests)
in some test there were 2 phones (Galaxy S3 and S7). S3 Locked but S7 not locked
in some other no phone locked but all GPS (in GPS Test android software) show 10 or more GPS signals and all were very good signal strength (45 to 50)

also i did some test with DJI Phantom 4 

I could spoof it in home but not outside(without amplifier)
Also could spoof my phone but could not spoof Mavic too (outside)

and ...

Now i have some guess and want to see if they are true ?

1- If signal strength goes up (35 to 50) some kind of saturation make spoofing incorrect.
2- Based on distance to our goal (Phone or Drone) we need some kind of amplifier to make signal correct level (Not below of real GPS signals and not so up to saturate)
3- Telescopic antenna may not good for this purpose
4- Why in some tests 1 phone lock but other not(both are in 1 meter distance)?
5- Why phone not lock even when they see 10 or more simulated sats ?
6- Please tell exact hardware config (Does attenuator needed ?)

any idea about above problems ?
Thnaks

osqzss, 2017-10-16:
Your question cannot be answered here because it involves intentional interference with the GPS signal.

EhsanArio, 2017-10-16:
Hi
What do you mean by this ? Do you mean your system works only somewhere if real gps signals are absent ?

If we amplify our gps signals our targets can see our gps signal ok ?

Solving interference is a software concept or communication concept?

Any solution ?

bitdust, 2017-10-16:
it's illegal.

EhsanArio, 2017-10-17:
In your country may be but in mine not

robellebor, 2017-10-19:
@osqzss "Your question cannot be answered here because it involves intentional interference with the GPS signal."  

Isn't that the main idea of this project ??? spoofing ???

Honky103, 2017-10-19:
The main idea is to create a cheap gps signal simulator for receiver testing, not to use as a spoofer

closed 2017-11-08 by osqzss