osqzss/gps-sdr-sim #89: How is eph2sbf used in generating the signal file (.bin)?

Honky103, 2017-10-16:
Hi,

I assume eph2sbf is used to generate the navigation message required to produce the signal file. I tried playing around with the binary messages in the subframes to see how does a u-blox-M8N respond. In particular, I only made subframe 5 a valid subframe (by using invalid subframe numbers for the other subframes) and make subframe 5 an all-zero packet (apart from the preamble and subframe ID). However, the receiver is still able to decode the timing information from it. My guess is the timing information is not only included in the navigation message generated by eph2sbf, and also in other places (e.g. ionoutc? maybe). Any guesses?

P.S I tried changing toe and toc values in subframe 1 to all-zeros yet it still showed the correct time. I did the experiment using a wired connection so all the receiver is receiving is from the signal simulator.

osqzss, 2017-10-16:
The signal transmission time is embedded into a part of HOW computed in generateNavMsg. TOE and TOC are the application times of the orbit and clock parameters in the broadcast ephemerides and not for the timing information.

Honky103, 2017-10-16:
Oh I completely missed that function! Thanks for your help! That also explains why the receiver outputs the correct WN despite it being set to 0 in eph2sbf.

closed 2017-10-16 by osqzss