osqzss/gps-sdr-sim #92: C/A codegen function

ongbuntu, 2017-10-25:
In
```
void codegen(int *ca, int prn)
{
...
for (i=0; i<N_DWRD_SBF; i++)
		r1[i] = r2[i] = -1;
...
}
```
why is initialization vector for G1 and G2 -1,-1,-1,-1,-1,-1,-1,-1,-1,-1 instead of 1,1,1,1,1,1,1,1,1,1?

fabriciocscte, 2017-10-25:
If you use multiplication in the calculation,  you should initialize the register with -1.
If you use module-2 or xor in the calculation, you should initialize the register with 1.
1  *   1    = 1
1  *  -1  = -1
-1 *  1   = -1
-1 * -1  = 1
-------------------------------------------
1  xor  1 = 0
1  xor  0  = 1
0  xor  1   = 1
0 xor   0  = 0
in this case you need to convert -1=>1  and 1 =>0
----------------------------------------------------------
Look below they use multiplication in the polymonials, so it should be initialized with -1

for (i=0; i<CA_SEQ_LEN; i++)
	{
		g1[i] = r1[9];
		g2[i] = r2[9];
		c1 = r1[2]*r1[9];
		c2 = r2[1]*r2[2]*r2[5]*r2[7]*r2[8]*r2[9];

		for (j=9; j>0; j--) 
		{
			r1[j] = r1[j-1];
			r2[j] = r2[j-1];
		}
		r1[0] = c1;
		r2[0] = c2;
	}

	for (i=0,j=CA_SEQ_LEN-delay[prn-1]; i<CA_SEQ_LEN; i++,j++)
		ca[i] = (1-g1[i]*g2[j%CA_SEQ_LEN])/2;

fabriciocscte, 2017-10-25:
The last for is just a conversion like this: -1=>1 and 1 =>0

ongbuntu, 2017-10-25:
Thank you for the clear explanation, fabriciocscte.

closed 2017-10-25 by ongbuntu