osqzss/gps-sdr-sim #93: User defined gps time

ongbuntu, 2017-10-27:
I am trying to accomplish the following:
1) define gps time manually and
2) use -t option to determine the satellites in view
if i use -T option,  it will mess up the satellites in view determined by the -t option. 
which part of the code should i focus on changing if i wish to accomplish the above?

Honky103, 2017-10-27:
Hi ongbuntu, not too sure what you are looking for specifically. You can use -T option to define gps time manually. And while generating the gpssim.bin file, it will output the satellites in view (its PRN and azimuth elevation information). These satellites in view differ when using the -t option and -T option because the gps time differs, hence the satellites in view is different.

ongbuntu, 2017-10-27:
Hi Honky103,
Thanks for the reply.
I can give an example to illustrate:
If I specify -t 2017/10/30,00:00:00, the gpssim.bin would be generated based on a certain sets of satellites in view computed using the ephmeris file.
If I specify -T 2017/10/30,22:00:00, the toc and toe would be overwritten with this value and the satellites in view computed with therefore be different.
What I want to accomplish is to use -t 2017/10/30,00:00:00 to compute the satellites in view, but to specify the spoofing start time manually, say, 2017/10/30,22:00:00. I want to know which part of the code I should change in order to achieve the above.

Hope that I am making some sense in my explanation.

Appreciate your advice, pse.

Honky103, 2017-10-27:
Okay I understand your requirement as this - For a specified location, use the satellites in view at 2017/10/30,00:00:00 to spoof the timing to be 2017/10/30,22:00:00, while indicating that the satellites are at the position they are supposed to be at 2017/10/30,00:00:00.

If that's the case, you have to change the toe of the RINEX ephemeris data. I have not done that before, but I believe it can be done if you change the tow value in the generateNavMsg function to the value that you want it to be (in this case it's the seconds of week for 2017/10/30,22:00:00), and use -t 2017/10/30,00:00:00. I think this will trick the simulator to search for satellites in view at 2017/10/30,00:00:00, but have its tow altered midway so your receiver receives 2017/10/30,22:00:00 timing.

ongbuntu, 2017-10-27:
Hi Honky103,

Thanks for the nice idea, I will try it and provide an update.
I am not good at programming so any idea that works would be good enough for me :-)

ongbuntu, 2017-10-27:
I tried the following:
`int generateNavMsg(gpstime_t g, channel_t *chan, int init)
{
...
	wn = (unsigned long)(g0.week%1024);
	tow = ((unsigned long)g0.sec+**3600**)/6UL;`

For some reasons, I am able to see that the satellites are in used and the receiver seems to be able to generate a satellite map for the SVs. However, I can never get a location fix and it seems that the receiver would go into reacquisition mode after a while :-(
I also tried changing the sec offset value to a small value say, 600. The result is the same. Could there be some other places where the time needs to be offset as well?

closed 2017-10-30 by ongbuntu

Honky103, 2017-11-01:
I'm not too sure why. It could be from receiver's point of view, when they get the TOW and TOE values as what you keyed in, the resulting pseudoranges computed do not converge to a positioning solution.