osqzss/gps-sdr-sim #94: About prolonging 0.1 second update interval

CollusW, 2017-10-30:
The gps-sdr-sim updates parameters such as pseudorange and Doppler shift every 0.1 seconds. I wonder why choosing 0.1 seconds? 

In fact, I tried to change this time interval to 0.2s to reduce the calculation load for some weak CPUs. However, the receiver may not get fixed in some tests or get lost (unfixed) for several seconds after fix.

I wonder this 0.1 seconds update interval cannot be prolonged theoretically, or it is due to my code bug? If latter case, could you help review my code? Many thanks!

osqzss, 2017-10-30:
During the update interval, it is assumed that the Doppler is constant and the pseudorange is linearly evolving. The shorter the update interval, the smaller the error. I chose 0.1 seconds to keep the pseudorange error small enough even in a high dynamics scenario. I think 0.2s should work for stationary antennas.

CollusW, 2017-11-01:
ok, thanks. I will do some experiments with the program to record the pseudorange and Doppler and see how fast it changes. I will come back with the results later.

CollusW, 2017-11-02:
Hi osqzss,

I modified the code to print out the pseudorange and Doppler to files (duration is set to 10s), read the values which changes every 0.1s and plotted with Matlab. 
I also checked the how much the pseudorange and Doppler changed during the 10s.
I tested under several time and location scenarios.

The detailed results are as follows.
1. pseudorange vs. time
[image]

2. Doppler vs. time
[image]

You can see they are almost flat during the 10s. For other test cases, the figures are almost the same and I do not show here, where the pseudorange and Doppler are almost flat.  Further, I find the max and min value for each PRN and calculate the percentage it changes ( (max-min)/min*100% ). The results are show in the table below. 

Table. Change rate in 10 second
[image]

some conclusions from the results:
During 10 seconds, 
1. The pseudorange changed within 7000m, the change rate is less than 0.05%;
2. Doppler changed within 10Hz, the change rate is less than 15%. (Note: worst case is test case #8 with 12%. However, this is due to the small Doppler value (~25Hz) )

Assuming linear change during the 10s, the doppler changes within 0.2s shall not exceed 0.3%. I think this is small enough to relax the update interval from 0.1s to 0.2s. And we can make it even longer so that some less powerful CPUs can calculate in real time.

What do you think?

CollusW, 2017-11-02:
Hi @osqzss ,

I upload the modified code to my gitHub (https://github.com/CollusW/gps-sdr-sim.git). It changed to 0.2s update interval. I believe it should work according to the above tests. However, as mentioned before, the receiver may not get fixed in some tests or get lost (unfixed) for several seconds after fix.

Could you pls help review the code and check if anything I missed?

Many thanks!

osqzss, 2017-11-02:
I took a quick look at your code. One thing I noticed that the navigation message was not renewed every 30 seconds with 0.2 update interval. In this case, the igrx counter of 150 is equal to 30 seconds.

CollusW, 2017-11-06:
I followed your advice and update the code. Then, the issue is fixed (receiver fixed successfully and no unstable issue occurs). Thanks!

btw, regarding the max update interval, the receiver's code tracking loop may get lost if the pseudorange changes too fast. Usually, the code loop requires the change of pseudorange within 1 C/A chip, which corresponds to about **0.4s** according to the simulation.

osqzss, 2017-11-12:
Thank you for your insight.

closed 2017-11-12 by osqzss