osqzss/gps-sdr-sim #100: Any way to increase dynamic Tx max from 300 to anything higher?

kpeezy707, 2017-11-04:
If so is there a max limit to what i can max it out at? lol

kpeezy707, 2017-11-04:
Update: Could i just change this line of code or would it break things?...

/*! \brief Maximum number of user motion points */
#define USER_MOTION_SIZE (3000) // max duration at 10Hz

Thanks

osqzss, 2017-11-07:
Yes, you can increase the size of USER_MOTION_SIZE.

kpeezy707, 2017-11-12:
I tried to increase this number in gpssim.h file. No change occurred, it still only went for about 5 minutes. Any where else i need to change something?

Thanks

kpeezy707, 2017-11-12:
I also notice when i build the gpssim.bin file from the NMEA file it stops at 300. "Time into run = 300.00, and it wont go any further than that

osqzss, 2017-11-12:
You need to prepare a longer user motion file and set the scenario duration with "-d" option.

kpeezy707, 2017-11-12:
Sorry to be a pain, i have made a longer user motion, i also used the -d option but it seems to ignore what i changed it to in the gpssim.h file. When using the -d optinon i get "ERROR: Invalid Duration" until i set it to 300 even if in the .h file its max is defined as 300000. Hope that makes sense.

osqzss, 2017-11-12:
I was unable to replicate the issue. I could extend the scenario duration time by simply increasing the value of USER_MOTION_SIZE.

[image]

kpeezy707, 2017-11-13:
Well shit lol i gotta be doing something wrong. Im on linux, do i need to rebuild the directory after changing code? Gonna keep trying. Thank you

kpeezy707, 2017-11-14:
This is line in gpssim.h to edit right?

/*! \brief Maximum number of user motion points */
#define USER_MOTION_SIZE (6000) // max duration at 10Hz

I have it set there for 6000

This is the line i use to make the bin file

./gps-sdr-sim -e brdc3070.17n -g DOPE2.txt -d 600

This is where i keep getting invalid duration, do those steps look correct to you?

osqzss, 2017-11-14:
what is the content of your DOPE2.txt file? The NMEA option requires 10Hz GGA data stream. Take a peak into triumphv3.txt in the repository as an example.

kpeezy707, 2017-11-15:
I was finally able to build and run on my windows box...i changed the code and got the dynamic mode duration to succesfully change to 600 like in your provided example. Which means im doing something wrong on my side.

Im running Kali linux. When i make a code change in linux to the gpssim.h file do i need to do some kind of rebuild or make or something?

Thanks a lot for your help

kpeezy707, 2017-11-15:
Also, the highest i could set the USER_MOTION_SIZE to is 34000, if i do 35000 and build the project the program will not run and go to "not responding". May be a bug

osqzss, 2017-11-15:
I'm not a Linux person, but as far as I know you need to specify the dependencies in the makefile rules to rebuild the object files when the header files change.

The USER_MOTION_SIZE limit seems somehow related to the size of 16-bit signed integer, but it's still incomprehensible why it works with 34000.

kpeezy707, 2017-11-16:
Gotcha, ill be looking into that, i aint a linux guy either lol.

Gotcha!, lol ey it works though

Is there anything special i need to do to get the blade to run on windows? like run that blade script. haven't done that one before...Im so used to apt get.

Thanks

closed 2017-11-23 by osqzss