osqzss/gps-sdr-sim #103: baseband or IF signal

fabriciocscte, 2017-11-24:
I have a question. Does it generate a baseband signal or an intermediate frequency signal? Does the signal digitized signal in a low frequency?

osqzss, 2017-11-24:
It generates digitized zero-IF I/Q baseband signals.

fabriciocscte, 2017-11-27:
I know it's a conceptual deficiency of mine, but I still have a question.

I use if signals like this:  IntermediateFreq: 3.563MHz with FS=11.999Mhz  for instance.
In my acquisition algorithm, I put these two values and I get the correlation peak.

Can you give an example what I need to put if I use a signal generated from gps-sdr-sim?

osqzss, 2017-11-27:
If you want to process the baseband signal generated by the gps-sdr-sim, your code acquisition algorithm has to handle the zero-IF I/Q signal. I think GNSS-SDR mailing list is a better place to post your receiver-related question: https://sourceforge.net/p/gnss-sdr/mailman/gnss-sdr-developers/

closed 2017-12-11 by osqzss