osqzss/gps-sdr-sim #104: Combining gps-sdr-sim with GNSS-SDR

robellebor, 2017-11-24:
In order to generate a navigation file with a current RINEX file instead off past RINEX files downloaded from sites, i think its better to combine it with "https://github.com/gnss-sdr/gnss-sdr" ... 
what gnss-sdr does is, it receives signal from the satellite and it gives us organized RINEX files ,  (gps_ephemeris.xml file, .17N file, .17O file, nmea_pvt.nmea file, .geojson file, .kml file and .rtcm file) so the gps-sdr-sim can read this recent files and generate a signal similar to the signals that are generated by the satellite at that moment. 

All we have to do is save the GNSS-SDR "receiver.conf" file inside the gps-sdr-sim directory and run it...

osqzss, 2017-11-25:
We welcome your contribution.

Fles42, 2018-03-06:
hi, 

how do you use RINEX data from GNSS-SDR to GPS-SIM-SDR? I tried, but RINEX from ftp://cddis.gsfc.nasa.gov/gnss/data/daily/ is version 2 and file from GNSS-SDR is version 2.11. During the test i received the meesage: ERROR: no current set of ephemerides not found.

Thanks, Tomáš

osqzss, 2018-03-07:
GPS-SDR-SIM should be capable of reading RINEX GPS navigation message data in version 2.11 format. ftp://igs.org/pub/data/format/rinex211.txt

Would you show me an example RINEX navigation file generated by GNSS-SDR?

Fles42, 2018-03-07:
Thanks for your respoponse.

Here is a example RINEX fire from GNSS-SDR from yesterday:

     2.11           N: GNSS NAV DATA    G: GPS              RINEX VERSION / TYPE
GNSS-SDR            ras                 20180306 150511 UTC PGM / RUN BY / DATE 
GPS NAVIGATION MESSAGE FILE GENERATED BY GNSS-SDR           COMMENT             
GNSS-SDR VERSION 0.0.9.git-next-e352978                     COMMENT             
See http://gnss-sdr.org                                     COMMENT             
GPSA    .0000D+00   .0000D+00   .0000D+00   .0000D+00       IONOSPHERIC CORR    
GPSB    .0000D+00   .0000D+00   .0000D+00   .0000D+00       IONOSPHERIC CORR    
GPUT   .0000000000D+00  .000000000D+00      0 1024          TIME SYSTEM CORR    
     0     0     0     0                                    LEAP SECONDS        
                                                            END OF HEADER       
G12 2018 03 06 16 00 00  .346813816577D-03 -.204636307899D-11  .000000000000D+00
      .120000000000D+02 -.429375000000D+02  .401945314064D-08  .736296738560D+00
     -.224076211452D-05  .704509869684D-02  .618956983089D-05  .515363424492D+04
      .230400000000D+06 -.800937414169D-07  .300103899846D+01  .428408384323D-07
      .987519827467D+00  .274500000000D+03  .931035940150D+00 -.802783439153D-08
     -.411088552060D-09  .100000000000D+01  .199100000000D+04  .100000000000D+01
      .000000000000D+00  .000000000000D+00 -.125728547573D-07  .120000000000D+02
      .227118000000D+06  .400000000000D+01                                      
G19 2018 03 06 16 00 00 -.452678650618D-03  .261479726760D-11  .000000000000D+00
      .490000000000D+02 -.173750000000D+02  .383051669925D-08  .821442094943D+00
     -.791624188423D-06  .973077991512D-02  .109001994133D-04  .515370693779D+04
      .230400000000D+06 -.175088644028D-06 -.220100938675D+01 -.819563865662D-07
      .979610177587D+00  .185812500000D+03  .113187242922D+01 -.781889711702D-08
      .328585115460D-10  .100000000000D+01  .199100000000D+04  .100000000000D+01
      .000000000000D+00  .000000000000D+00 -.153668224812D-07  .490000000000D+02
      .227118000000D+06  .400000000000D+01

Fles42, 2018-03-07:
[GSDR065q04.zip](https://github.com/osqzss/gps-sdr-sim/files/1788694/GSDR065q04.zip)

alkuna, 2018-03-07:
This is not a RINEX 2.11 file, but RINEX 3.01 with wrong header. See ftp://igs.org/pub/data/format/rinex211.txt and https://kb.igs.org/hc/en-us/article_attachments/115007741928/rinex301.pdf.
Support for RINEX 3.x is almost done. Give some more time please.

osqzss, 2018-03-07:
It clearly violates the version 2.11 format in may ways. For example, the satellite PRN number has to be a two-digit integer without 'G' character, and the year value of TOC has to be a two-digit as well. GPS-SDR-SIM cannot recognize any non-standard input files.

Fles42, 2018-03-07:
Hi, thanks fo your answers.

I changed some parameters in GNSS-SDR for RINEX v2.11 and now I get, I think, correctly file. But I still get the same error message. It is beceuse zeros in ION ALPHA, ION BETA and DELTA-UTC: A0,A1,T,W?

     2.11           N: GPS NAV DATA                         RINEX VERSION / TYPE
GNSS-SDR            ras                  7-MAR-18 13:25     PGM / RUN BY / DATE 
GPS NAVIGATION MESSAGE FILE GENERATED BY GNSS-SDR           COMMENT             
GNSS-SDR VERSION 0.0.9.git-next-e352978                     COMMENT             
See http://gnss-sdr.org                                     COMMENT             
     .0000D+00   .0000D+00   .0000D+00   .0000D+00          ION ALPHA           
     .0000D+00   .0000D+00   .0000D+00   .0000D+00          ION BETA            
     .000000000000D+00  .000000000000D+00        0     1024 DELTA-UTC: A0,A1,T,W
     0                                                      LEAP SECONDS        
                                                            END OF HEADER       
 1 18  3  7 14  0  0.0 -.327280722558D-04 -.227373675443D-11  .000000000000D+00 
     .580000000000D+02  .112500000000D+01  .444518515980D-08  .100087949819D+01 
    -.111758708954D-07  .740886246786D-02  .749528408051D-05  .515367296982D+04 
     .309600000000D+06  .204890966415D-07 -.124547757679D+01  .819563865662D-07 
     .970108211598D+00  .239843750000D+03  .610427899001D+00 -.813605318499D-08 
     .315370279295D-09  .100000000000D+01  .199100000000D+04  .100000000000D+01 
     .000000000000D+00  .000000000000D+00  .558793544769D-08  .580000000000D+02 
     .307488000000D+06  .400000000000D+01

[GSDR066o23.zip](https://github.com/osqzss/gps-sdr-sim/files/1789890/GSDR066o23.zip)

alkuna, 2018-03-07:
I had no problem processing the file, see log.txt
Still, the file is not perfect. There is a spare space between the columns, but it just reduces the resolution.
I also attach your RINEX file fixed by [teqc](https://www.unavco.org/software/data-processing/teqc/teqc.html).

[log.txt.gz](https://github.com/osqzss/gps-sdr-sim/files/1789996/log.txt.gz)
[GSDR066o23.18N.gz](https://github.com/osqzss/gps-sdr-sim/files/1790000/GSDR066o23.18N.gz)

closed 2018-03-21 by osqzss

DarshanNathvani, 2023-05-23:
> This is not a RINEX 2.11 file, but RINEX 3.01 with wrong header. See ftp://igs.org/pub/data/format/rinex211.txt and https://kb.igs.org/hc/en-us/article_attachments/115007741928/rinex301.pdf. Support for RINEX 3.x is almost done. Give me some more time please.

If Your Work Is Done Please Share Updated Code