osqzss/gps-sdr-sim #109: C/A code generation...code chip delay

bikilay, 2018-01-09:
in ca code generation, when code selection takes place how 2 XOR 6 form code chip delay of 5, and 3 XOR 7 forms 6 etc....how these delays created??

osqzss, 2018-01-09:
You don't need the both. The delay in the G2 generator is equivalent to the XOR of the selected taps.  You can implement the code generator in either way.

bikilay, 2018-01-10:
that is ok..but i don not understand how delay in G2 generator formed...how 2 XOR 6 form 5 delay..??
    my point is how these delays are formed with code selections??

osqzss, 2018-01-10:
I don't know the exact way to convert a given phase delay into a pair of XOR taps, but it is known as "shift and add" property of any m-sequence. The different pairs of XOR taps yield the same m-sequence with different delays.

bikilay, 2018-01-10:
ok thanks

closed 2018-01-11 by osqzss