osqzss/gps-sdr-sim #113: I Can't Run the Problem correctly.

woshiluozhibin, 2018-01-14:
Hello,
    I meet a problem here, the parameter argc is one by default. And the problem would jump to usage and exit function. Can you please give me some guidance about how to use this program?

    Thank you very much.

osqzss, 2018-01-14:
You need to at least to specify the name of a RINEX navigation file.

woshiluozhibin, 2018-01-15:
Thanks for your reply!
I don't know hwo to specify the name of RINEX navigation file. I attempt to do some revisions as you suggested. But they don't work.
(1) The value of variable "navfile[0]" is set to the path of navigation file (including the file name).
(2) The value of staticLocationMode is set to be TRUE.
(3) It should be noted that we don't use the precompiled header because it would cause some erors. But it seems like the variable definition parts of main function are not executed. 
(4) Should the variable "argc" be assigned to a specified value before executing the program?

osqzss, 2018-01-15:
gps-sdr-sim is a command line program. Did you follow the "Generating the GPS signal file" instructions in the README.md?

woshiluozhibin, 2018-01-15:
Sorry, I am a beginner in the aspect of VS. But after I bulit the program, the generated .exe file in "Release" folder cannot be executed. I think I should run this .exe file to using the command line. Is it because I cancel the  precompiled header? But if I don't canceal it, the VS would indicates an error. (error C1853, 'Release\IF_Sig_Sim.pch' precompiled header file is from a previous version of the compiler, or the precompiled header is C++ and you are using it from C (or vice versa) ).

osqzss, 2018-01-15:
Did you create an empty C/C++ project for a console application on VS?
https://msdn.microsoft.com/en-us/library/ms235629.aspx

woshiluozhibin, 2018-01-15:
Thanks for your careful instruction.

Indeedly, the error C1853 occurs because I didn't select the "empty project" when I create the Win32 Console Application. This problem has been solved. 

When I press the "ctrl + F5" buttons in "Release" mode, the following window is appeared.
[image]

If I run the generated ".exe" file in "Release" folder, a window would flash and disappear. In such a condition I cannot type any commands in this window.  Are there any operations that I ignored?

I don't know if making the responses would impact the effectiveness of your work. Please make the reply at your convenience. 

Thank you very much.

osqzss, 2018-01-15:
You have to run the code from the command prompt. The easiest way to open the command prompt windows is to type "cmd" into the address bar of the Release folder and press Enter.

[image]

Then you need to type the command and the options into it.

[image]

woshiluozhibin, 2018-01-15:
I followed your instructions and generated the IF samples successfully.  GNSS-SDR can processing the IF samples  and eventually calculate the PVT parameters of GNSS receiver. 

This project provides a low-cost solution for Software/Hardware GNSS signal simulator.  It is really a powerful tool. 

ありがとうございます。^ ^

closed 2018-01-15 by osqzss